absalom-muc/MHI-AC-Ctrl #78: Possible to add PIR function?

sacy80, 2022-01-29:
Hi thank alot for the work. i try and it work. but i not good at coding. would like to know possible to add pir as a binary sensor to sent mqtt messages ?

absalom-muc, 2022-01-29:
short answer: Yes, should be possible

Usually you find in the web examples for many different sensors connected to ESP8266 and used with MQTT

sacy80, 2022-01-30:
ar, ao ijust add everyting in void loop ?  very sorry to ask about this.

absalom-muc, 2022-01-31:
yes, in the loop() you read your PIR status and you publish the status with e.g. 
`MQTTclient.publish("topic", "data", true); `
Please check the PubSubClient documentation for details

sacy80, 2022-02-02:
ok. noted. i try to figure out.. thanks alot for info and help :)

sacy80, 2022-02-02:
[image]
Thanks alot for the info :) look like working :)

closed 2022-02-02 by sacy80

absalom-muc, 2022-02-02:
you could simplify the code:
replace
[image]
by
[image]
Since you are doing the same in both branches you don't need the if-clause

sacy80, 2022-02-03:
oh.. i try. .. thanks for the help :)