absalom-muc/MHI-AC-Ctrl #79: Provide Wifi and MQTT info after flashing

WorlockM, 2022-02-17:
Before I found this project, I tried to get my airco on WiFi with the [MHI2MQTT](https://github.com/rjdekker/MHI2MQTT) project from rjdekker. Now that project didn't work at all for me, so I am very glad I found this one. But one thing I did like about his method, is that when you connect the module, it first becomes a hotspot, and you van connect to it, and then enter your WiFi and MQTT information.

This makes it easier if you want to make this module for someone else. Maybe it's an idea to add this functionality?

absalom-muc, 2022-02-20:
This was already implemented by @wickeddurkie, you find it [here](https://github.com/absalom-muc/MHI-AC-Ctrl/pull/69).

absalom-muc, 2022-06-16:
closed since there was no update for a long time. Please feel free to re-open in case of follow-up topics.

closed 2022-06-16 by absalom-muc

DYLaKo, 2022-06-16:
Hi @absalom-muc, the implementation of WickedDurkie is in an older version of your code. Is there any change you might reconsider adding this to your own code?