absalom-muc/MHI-AC-Ctrl #80: mhi_ac_ctrl_core.loop error: -3

andi377, 2022-02-23:
I described the wemos with version v2.5R21. The Wemos is available in the WiFi and also in the MQTT server in the iobroker. The data points have been created, but they are not updated. Where I have now activated the serial monitor in the IDE comes

18:01:26.323 -> mhi_ac_ctrl_core.loop error: -3
18:01:26.423 -> mhi_ac_ctrl_core.loop error: -3
18:01:26.523 -> mhi_ac_ctrl_core.loop error: -3
18:01:26.624 -> mhi_ac_ctrl_core.loop error: -3
18:01:26.724 -> mhi_ac_ctrl_core.loop error: -3

Where is my problem, unfortunately I can't get any further

absalom-muc, 2022-02-23:
This message indicates a time out condition for the SCK pin. Please provide the debug outputs from the beginning after reset is released. There should be messages related to the pins SCK, MISO and MOSI, so that you can see if the pins are connected in the right way. (I assume SCK is not connected, but let's see)

andi377, 2022-02-24:
Ifound the problem, it was a hardware error, now everything works.
Thanks for your help

Is there also a possibility to integrate a ceiling device (FDTC25VF) ?

andi377, 2022-02-25:
Today I wanted to rewrite the wemos, only the host name was adjusted. After that, the control no longer works.

The  debugfile is attach 

Here is the problem?

[debug.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/8142606/debug.txt)
-

andi377, 2022-02-26:
yes it the original HW setup for MHI-AC-Ctrl).
I just changed the hostname.
With one pcb I have the plug directly with the second one with a cable length of approx. 15 cm. I measured the pcb electrically. Only the level converter could be the problem
.
In the MQTT server I can see that the time stamp for MISO, MOSI and SCK changes when the Wemos is boote
[debug2.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/8147493/debug2.txt)
[image]
[image]
[image]
d

absalom-muc, 2022-02-28:
The frequency values for SCK, MOSI and MISO during boot look perfect and I can see in your debug file that it works without problems for 20 seconds. I have no idea what happens after these 20 seconds. Maybe the esp8266 should be replaced, but that's just a guess.

andi377, 2022-03-03:
oday I tried it again with new hardware,
one version worked, one didn't.
When it worked, the whole thing only went for a certain amount of time. Then the wemos was no longer visible in the WLAN. When checked, the wemos was very warm. What temperature is normal there?
I'm slowly running out of ideas on how to get this to work.

absalom-muc, 2022-03-12:
> oday I tried it again with new hardware, one version worked, one didn't. When it worked, the whole thing only went for a certain amount of time. Then the wemos was no longer visible in the WLAN. [...]

I don't know "normal temperature".
Not sure what you mean with "one version worked, one didn't". Have you exchanged the Wemos of the boards? So not clear if you identified a board not working or a Wemos not working.

absalom-muc, 2022-06-05:
Since there was no update since long time, I assume this issue is resolved. In case of further questions please don't hesitate to re-open it.

closed 2022-06-05 by absalom-muc