absalom-muc/MHI-AC-Ctrl #81: Idea to improve precision

ph-pk, 2022-02-23:
The MHI units have a precision of 1.0C for setpoint. Additionally, troom is rounded down to .0 / 0.25 / .5 / .75 causing an additional loss of precision for unit control logic when e.g. an external sensor with MQTT is used.

To improve this behavior I prototyped the following:
- properly round set/Tsetpoint to nearest integer
- store difference as offset
- correct set/Troom with offset
- properly round corrected troom to nearest 0.25
This emulates a precision of 0.25C, max difference 0.125C.

Would this be something to include?

kARTechnology, 2022-02-25:
i think 0.5 deg precision is possible as in eco mode it is using set temp=set temp+1.5deg 
anyone with wired remote can tell if it is possible to set in 0.5 deg steps

kARTechnology, 2022-02-25:
https://github.com/absalom-muc/MHI-AC-Ctrl/issues/74

absalom-muc, 2022-02-26:
> The MHI units have a precision of 1.0C for setpoint. Additionally, troom is rounded down to .0 / 0.25 / .5 / .75 causing an additional loss of precision for unit control logic when e.g. an external [...]

Thanks for your suggestion @ph-pk.

Not sure that I understand it completely. Some comments:

- The resolution (not the precision) for the setpoint was 1°C so far, but it changed with v2.5R21 to 0.5°C (btw: I missed to mention this in [Version.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Version.md))
- Troom resolution is 0.25°C. Also when you send the room temperature via MQTT then MHI-AC-Ctrl writes this value (with the resolution of 0.25°C) to the AC unit. But only changes>=0.5°C will be published as discussed in your issue #82. 
- The measurement of the room temperature has according to my experience usually an accuracy >1°C. But I don't see that as a problem since the human temperature sense is not stable. Using a higher resolution would only counterfei an accuracy for the temperature measurement and control that is not given.

That's why I'm rather reluctant when it comes to supposed ideas for increasing accuracy.

pkoster, 2022-03-02:
(issue was created with wrong account)

> * The resolution (not the precision) for the setpoint was 1°C so far, but it changed with v2.5R21 to 0.5°C (btw: I missed to mention this in [Version.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Version.md))

I performed an experiment to validate this. Logged values to Influxdb and visualized with Grafana.
The findings contradict with 0.5°C and confirm 1.0°C resolution:
- the unit switches down/off when troom/op_returnair reaches op_tsetpoint. The unit switches up/on when troom/op_returnair falls below op_tsetpoint.
- op_tsetpoint has a resolution of 1.0°C (setpoint/set_setpoint + 2.0 rounded up)
- the resulting temperature subsequently centers around op_tsetpoint

Example: set_setpoint=18.5°C leads to setpoint=18.5°C with op_setpoint=21.0°C and also a resulting temperature around 21.0°C.

Conclusion: it is possible to specify setpoint with 0.5°C resolution but the effective resolution is 1.0°C.

[image]

> * Troom resolution is 0.25°C. Also when you send the room temperature via MQTT then MHI-AC-Ctrl writes this value (with the resolution of 0.25°C) to the AC unit. But only changes>=0.5°C will be published as discussed in your issue [MQTT Troom update on smallest change #82](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/82).

Will get back to this separately.

> * The measurement of the room temperature has according to my experience usually an accuracy >1°C. But I don't see that as a problem since the human temperature sense is not stable. Using a higher [...]

Humans (and some thermometers) are indeed bad at measuring absolute temperature, but pretty good at sensing relative temperature differences. Particularly for people, 0.5°C difference can make a real-world difference between comfortable and non-comfortable.

Therefore, I'd argue to at least try to support 0.5°C (and more where possible).

Below is a quick & dirty example to achieve a 20.5°C temperature.
[image]

absalom-muc, 2022-03-12:
Thanks for the explanation and the diagrams, I see the reduced variation of set_troom.

Some topics are still unclear for me:
- I assume troom is hiden by op_returnair because they are equal, please confirm.
- Why is set_setpoint (red) not equal to setpoint (dark blue) in the second diagram? In the first diagram they are equal.

To be honest I can still not comprehend that the 0.5°C should make a big difference, but I accept that other persons might feel different.

Have you implemented your changes in MHI-AC-Core-ctrl.cpp only or are other files affected, too?

kARTechnology, 2022-05-18:
Hi @ph-p

> The MHI units have a precision of 1.0C for setpoint. Additionally, troom is rounded down to .0 / 0.25 / .5 / .75 causing an additional loss of precision for unit control logic when e.g. an external [...]

Hi, can you share the changes made in code to get more tighter control of the set temperature?

Also is actual set temperature = tsetpoint-2 degree c?

rwijnhov, 2022-06-09:
@pkoster could you post your changes?

pkoster, 2022-06-11:
@absalom-muc : the diagrams were made with slight differences in code that did / did not correct MQTT values like setpoint (dark blue) for the offset.

@kARTechnology : the code below does not incorporate the -2 offset 

@rwijnhov : 
Below is a unified diff, which has been subsequently manually stripped of configuration, non-working extensions, etc. (untested).

`diff -U 5 ../MHI-AC-Ctrl/src/MHI-AC-Ctrl-core.cpp MHI-AC-Ctrl/MHI-AC-Ctrl-core.cpp
--- ../MHI-AC-Ctrl/src/MHI-AC-Ctrl-core.cpp	2022-01-09 15:01:48.503850946 +0100
+++ MHI-AC-Ctrl/MHI-AC-Ctrl-core.cpp	2022-03-05 00:28:00.539499083 +0100
@@ -139,17 +139,41 @@
 
 void MHI_AC_Ctrl_Core::request_ErrOpData() {
   request_erropData = true;
 }
 
+float MHI_AC_Ctrl_Core::get_troom_offset() {
+  return Troom_offset;
+}
+
+void MHI_AC_Ctrl_Core::set_troom_offset(float offset) {
+  Troom_offset = offset;
+}
+
+int MHI_AC_Ctrl_Core::get_opdata_tsetpoint() {
+  return opdata_tsetpoint;
+}
+
+int MHI_AC_Ctrl_Core::get_status_tsetpoint() {
+  return status_tsetpoint;
+}
+
 #ifndef ROOM_TEMP_IU
 void MHI_AC_Ctrl_Core::set_troom(byte troom) {
   Serial.printf("MHI_AC_Ctrl_Core::set_troom %i\n", troom);
   new_Troom = troom;
 }
@@ -294,17 +318,18 @@
       }
       status_vanes_old = vanestmp;
     }
 
     int8_t troom_diff = MOSI_frame[DB3] - status_troom_old; // avoid using other functions inside the brackets of abs, see https://www.arduino.cc/reference/en/language/functions/math/abs/
-    if (abs(troom_diff) > 1) { // Room temperature delta > 0.25°C
+    if (abs(troom_diff) > 0) { // Room temperature delta > 0.0°C
       status_troom_old = MOSI_frame[DB3];
       m_cbiStatus->cbiStatusFunction(status_troom, status_troom_old);
     }
 
     if (MOSI_frame[DB2] != status_tsetpoint_old) { // Temperature setpoint
@@ -545,7 +571,19 @@
         break;
       default:    // unknown operating data
         m_cbiStatus->cbiStatusFunction(opdata_unknwon, MOSI_frame[DB10] << 8 | MOSI_frame[DB9]);
     }
   }
+  
+  if (Troom_offset != Troom_offset_old) {
+	  Troom_offset_old = Troom_offset;
+	  m_cbiStatus->cbiStatusFunction(troom_offset, (int)(Troom_offset*1000.0));
+  }
+  
   return call_counter;
 }
diff -U 5 ../MHI-AC-Ctrl/src/MHI-AC-Ctrl-core.h MHI-AC-Ctrl/MHI-AC-Ctrl-core.h
--- ../MHI-AC-Ctrl/src/MHI-AC-Ctrl-core.h	2022-01-09 15:01:48.503850946 +0100
+++ MHI-AC-Ctrl/MHI-AC-Ctrl-core.h	2022-03-02 23:19:10.322983463 +0100
@@ -61,10 +61,11 @@
   type_status = 0x40, type_opdata = 0x80, type_erropdata = 0xc0
 };
[... 149 lines cut ...]
+++ MHI-AC-Ctrl/support.h	2022-03-02 23:21:23.385894871 +0100
@@ -30,22 +30,25 @@
                                                     // enter 0 if you don't use the DS18x20 
 #define ONE_WIRE_BUS 4                              // D2, PIN for connecting temperature sensor DS18x20 DQ pin
 
 // Per default AC internal sensor is used for the room temperature. You can select using the DS18x20 sensor or providing Troom via MQTT.
 //#define ROOM_TEMP_DS18X20                         // use room temperature from DS18x20
-//#define ROOM_TEMP_MQTT                            // use room temperature from received MQTT topic
-#define ROOM_TEMP_MQTT_TIMEOUT  20                  // only considered if ROOM_TEMP_MQTT is defined
+#define ROOM_TEMP_MQTT                            // use room temperature from received MQTT topic
+#define ROOM_TEMP_MQTT_TIMEOUT  300                  // only considered if ROOM_TEMP_MQTT is defined
                                                     // time in seconds, after this time w/o receiving a valid room temperature
                                                     // via MQTT fallback to IU temperature sensor value
 
 //#define POWERON_WHEN_CHANGING_MODE true           // uncomment it to switch on the AC when the mode (heat, cool, dry etc.) is changed
                                                     // used e.g. for home assistant support
 
+#define ENHANCED_RESOLUTION true
+
 // *** The configuration ends here ***
                                                     
 #include <ESP8266WiFi.h>        // https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi
 #include <PubSubClient.h>       // https://github.com/knolleary/pubsubclient
 #include <ArduinoOTA.h>         // https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA
`

absalom-muc, 2022-08-05:
Not sure if you plan a pull-request, else please consider to close the issue

rwijnhov, 2022-09-11:
Could we please incorporate the .5 temp settings in the base code? I use external sensors. And 22 is too cold and 23 too warm :) 22.5 is the temp I need.

absalom-muc, 2022-09-11:
> Could we please incorporate the .5 temp settings in the base code? I use external sensors. And 22 is too cold and 23 too warm :) 22.5 is the temp I need.

Tsetpoint resolution of 0.5°C is already implemented since December 2021, please see https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md.

For the precision improve, someone has to initiate a pull-request, I don't plan it currently.

rwijnhov, 2022-09-11:
Yes I know but it doesn't work. If i set it to 22.5 it will still heat to 23 degrees same behaviour if inset it to 23.

glsf91, 2022-12-09:
I have made some changes so the setpoint of .5 degrees are working, based on the diff pkoster has supplied. But I made some changes to make it also work with a connected external DS18B20. Also when the wemos is reset or the remote control is used.
Basically the Troom send to the airco will be increased with an offset (0.5 degrees) when a setpoint x.5 degrees is used. 
I have to test this more, but I will make a pull request when finished.

glsf91, 2022-12-11:
Made a pull request for changing Tsetpoint resolution as discussed above. I used the diff supplied by @pkoster (thanks !) but changed a few things to make it work also after a reset of the wemos and when using the remote control.
Tested a lot with external sensor but also with set/Troom MQTT.

absalom-muc, 2023-01-28:
implemented in v2.6

closed 2023-01-28 by absalom-muc