absalom-muc/MHI-AC-Ctrl #82: MQTT Troom update on smallest change

ph-pk, 2022-02-23:
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/9491f193ab9f9ac87348584c15385fd71fdc4ade/src/MHI-AC-Ctrl-core.cpp#L299

The current code seems to update MQTT Troom only for >=0.5C changes. Other parameters seem to cause an update regardless of the size of a change.

Consequently, should/could 
`if (abs(troom_diff) > 1) { // Room temperature delta > 0.25°C`
be changed to
`if (abs(troom_diff) > 0) { // Room temperature delta > 0.25°C`

absalom-muc, 2022-02-26:
By intention the MQTT message is only updated for changes > 0.25°C. Originally every change was supported, but this led to a wobbling of the room temperature.

absalom-muc, 2022-08-05:
I believe the issue was addressed, therefore closed. If there are further questions, please feel free to re-open it.

closed 2022-08-05 by absalom-muc

absalom-muc, 2022-12-10:
Imagine as an example you have a room temperature of 21.5°C. The resolution of your measurement is 1°C. Let's assume the ADC (analogue to digital converter) converts
analogue input temperature<21.5°C to digital output 21°C and
analogue input temperature>=21.5°C to digital output 22°C 
If the real temperature is 21.5°C you will have the effect that by noise on the temperature signal and/or inside the ADC  it will be sometimes <21.5°C and sometimes >21.5°C. This leads to a toggle of the digital output signal between 21°C and 22°C. In reality I saw multiple toggles per second of the room temperature and found it annoying because of the increased MQTT load and the nervous display. Therefore I introduced the 'filtering'.

If you change [row 231 of MHI-AC-Ctrl-core.cpp](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/67ffef767d02bf76571c9f28944cb357ee688892/src/MHI-AC-Ctrl-core.cpp#L231) from
`if (abs(troom_diff) > 1) { // Room temperature delta > 0.25°C`
to
`if (abs(troom_diff) > 0) { // Room temperature delta > 0°C`
then you have the full accuracy with the potential disadvantage of a restless digital output.

glsf91, 2022-12-10:
I think you are right when indeed measuring a temperature sensor.
But in this case it is filtering the Troom reported by the airco. So just reporting what received from the airco. Now it is not always showing the right value.
If there is any wobbling then this should be fixed at the input of the Troom (so from external sensor or set topic Troom). I think :-). 
I have indeed changed the 1 to 0 and for me it is working fine :-)

glsf91, 2022-12-11:
Made a pull request to give the user the choice to use smaller changes for Troom.

absalom-muc, 2022-12-21:
implemented in 2.6R3