absalom-muc/MHI-AC-Ctrl #83: Question

mriksman, 2022-03-08:
Hi,

Apologies, this isn't an issue related to your program, but I have used your documentation and code to recreate this in ESP-IDF using the SPI driver and hardware timers to get accurate timings on the 40ms pause between frames.

With the MISO pin _disabled_ (configurable on the SPI driver), I get great data.
```
packet: 160    6c 80 04   48 00 2c 9a 00 00 88 00 ff ff ff ff ff 00 00   07 81    checksum: 1921 calc_chk: 1921 
```

As soon as _enable_ the MISO pin, **but do not send any data** (I leave the buffer `NULL`), I get completely different data.
```
packet: 33557    ec 00 09   20 09 b0 28 05 00 40 04 f8 ff ff ff ff 07 00   e0 00 (calc 57344)  rx_checksum: 1850
packet: 33558    6c 00 09   20 09 b0 28 05 00 40 04 f8 ff ff ff ff 07 00   e0 00 (calc 57344)  rx_checksum: 1722
packet: 33615    9c 01 24   80 24 c0 a4 14 00 00 11 e0 ff ff ff ff 1f 00   80 03 (calc 32771)  rx_checksum: 2025
packet: 33616    6c 00 09   20 09 b0 28 05 00 40 04 f8 ff ff ff ff 07 00   e0 00 (calc 57344)  rx_checksum: 1722
packet: 33826    6c 00 09   20 09 b0 28 05 00 40 04 f8 ff ff ff ff 0f 00   c0 01 (calc 49153)  rx_checksum: 1730 
packet: 34312    6c 00 13   40 12 60 31 05 00 40 04 f8 ff ff ff ff 07 00   e0 80 (calc 57472)  rx_checksum: 1702 
packet: 34313    6c 00 09   20 09 b0 30 05 00 40 04 f8 ff ff ff ff 07 00   e0 80 (calc 57472)  rx_checksum: 1730 
packet: 34314    6c 00 09   20 09 b0 30 05 00 40 04 f8 ff ff ff ff 0f 00   c0 81 (calc 49281)  rx_checksum: 1738 
packet: 34869    6c 00 09   20 09 b0 30 05 00 40 04 f8 ff ff ff ff 07 00   e0 80 (calc 57472)  rx_checksum: 1730 
packet: 34963    9c 01 24   80 24 c0 c4 14 00 00 11 e0 ff ff ff ff 1f 00   80 03 (calc 32771)  rx_checksum: 2057 
```
I presume the pin is probably being held low (instead of high-impedance/not configured), and the AC is seeing 0x00..?

Is that expected? Is the MOSI data supposed to look like that until you have done the 'sync' pattern of 20 frames and alternate bit 2 of DB14?

mriksman, 2022-03-08:
It appears it is my problem. 

I am not sure why turning on the MISO causes the issues, but the problem goes away when I hook my scope up to SCL and MOSI. The internal pullups probably do nothing considering there is a TXS0104 level shifter. 

Need to redesign my PCB!

closed 2022-03-08 by mriksman

absalom-muc, 2022-03-12:
The MOSI data should be indendent from MISO. So it really looks like a HW problem, maybe a missing GND?