absalom-muc/MHI-AC-Ctrl #84: No / partial data published

Aesculapius, 2022-04-10:
Programmed the ESP with latest release from src, made sure to use PubSub 2.7 and to be secure in soldering etc.
Plugged it into my MHI-SRK50SW and the module boots up and connects to both WiFi and MQTT broker.

But I'm only seeing partial data and can't send commands... so something is wrong. Now what is the best approach to debug?

This is all I see:
[image]

absalom-muc, 2022-04-10:
Have you considered the Troubleshooting Guide, especially [section 'Pins not properly connected'](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-pins-not-properly-connected)?

Aesculapius, 2022-04-10:
Yeah was doing that. Serial console displays "mhi_ac_ctrl_core.loop error: -3", so I'm off checking pins, thanks for the headsup :)

Aesculapius, 2022-04-12:
Anything I could do to check this? Should I grab the multimeter to measure voltages or something?

absalom-muc, 2022-04-12:
This log 
```
SCK frequency=0Hz (expected: >3000Hz) out of range!
MOSI frequency=564Hz (expected: <SCK frequency) out of range!
MISO frequency=57Hz (expected: ~0Hz) out of range!
```
clearly shows that there is something wrong with the electrical connections. Before we start with the debugging, two questions:
1. Have you used the original schematic and PCB or have you changed something? In case of changes, please describe them in detail.
2. Have you directly plugged-in the PCB into the AC or do you use a cable for the connection?

Aesculapius, 2022-04-12:
yeah it's the original schematic and PCB without changes, v2.2. The unit is plugged in directly into the AC without cable in between. See attached pics
[image]
[image]
.

absalom-muc, 2022-04-12:
o.k., looks good. Let's start with the measurement. MHI-AC-Ctrl PCB should be plugged in into AC but ESP8266 board should be deattached. Please ensure that you don't touch inside the AC - it is dangerous!
1. Between 2 red marked pins GND and +5V should be 5V:
[image]

2. Between 2 red marked pins GND and 3V3 should be 3.3V:
[image]

3. Between 2 red marked pins GND and SCK (high) should be approx. 2.5V:
[image]

4. Between 2 red marked pins GND and SCK (low) should be approx. 1.7V:
[image]

If you have deviations please tell me the values.

Aesculapius, 2022-04-12:
Thanks for taking the time to troubleshoot.
I;ve got some weird readings. Some fluctuate a bit and might be influenced a little by me having to hold the dupont cables with my finger to the test-unit, as the location of the airco isn't.....ideal :-)

GND - 5V = 5.03V DC
GND - 3.3V = 0.25 - 0.38v DC
GND - SCKhigh = 2.3v DC
GND - SCKlow = 0.09 - 0.11v DC

Attached the bottom of the unit. I've had the AC on 230V but not powered on, that okay?

[image]

absalom-muc, 2022-04-12:
The values are reasonable. Of course we can't measure the 3.3V domain when the ESP8266 is not connected-my fault. It makes no difference if the AC is powered.
So please repeat the measurement for GND - 3.3V and GND - SCKlow with ESP8266 board attached.

Aesculapius, 2022-04-13:
Measured again with Wemos D1 Mini attached:

GND - 3.3 = 3,29v DC
GND - SCKlow = 1,06v DC
GND - SCKhigh = 1,26v DC

absalom-muc, 2022-04-13:
Looks not too bad.
You wrote already that you use the latest version from the repository, however can you confirm that the following lines in [MHI-AC-Ctrl-core.h](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/src/MHI-AC-Ctrl-core.h)
[image]
are not changed?
To be honest I have currently no idea what the reason for the bug could be and how to continue, but I will continue to think about it ...

Aesculapius, 2022-04-13:
yes, those are correct. 
[image]

Good to read that the voltages appear to be correct. Honestly I have too little electrotechnical background to understand where to look. Would some kind of debug program on the wemos help to troubleshoot?

absalom-muc, 2022-04-13:
Yes, it is not relevant. With the standard program you use, you have already a good test of the MHI-AC-Ctrl input pins SCK and MOSI with the integrated frequency measurement. Since the inputs signals seem to be o.k., you could try replacing ESP8266. I suppose you don't have an oscilloscope on hand right now.

Aesculapius, 2022-04-13:
To be complete, I measured the voltages on the AC PCB, from left to right with pin 5 as Ground:
1 = 12.46v
2 = 2.01v
3 = 3.45v
4 = 1.10v

I don't have an oscilloscope. I'll try to reflash the Wemos with the stable branch for now, just to be sure...

absalom-muc, 2022-04-14:
I rechecked the values and there is something wrong with the voltages. To have a clear common reference I use the pin numbering from the [schematic](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md#schematic) for X1 (different to your numbering)
1 - GND  : 0V
2 - MISO : toggling between 0V and 3V, but currently irrelevant because it is our ouput
3 - MOSI : 4.7V (you have 3.45V)
4 - SCK   : 4.7V (you have 2.01V)
5 - +12V : 12.4V

We should check why you have significant lower voltages for SCK and MOSI. It could be related to different multimeters, but maybe not. Can you measure the X1 voltages without the MHI-AC-Ctrl PCB plugged in?

Aesculapius, 2022-04-14:
Got to thank https://github.com/DYLaKo for his tip to take a look at his manual. It's in Dutch, but it brought me at the right points to check the flashing procedure. I redid everything and I think that in my original flashes, I missed some libraries: Didn't install the "ESP8266 Microgear" from Chavee Issariyapat (didn't knew I needed that), didn't install the onewire and dallastemperature one (as I thought I didn't need them since no connected external temp. sensor is being used).

But once I installed all those modules and reflashed the Wemos, the unit throws a lot more data on mqtt, and I can controll the unit through HomeAssistant and everything!

[image]

So I'm happy, thankful to you and such great community.

absalom-muc, 2022-04-14:
I'm glad to hear that it works! 

But I don't believe the context to the missing libraries.
"ESP8266 Microgear" is not used in MHI-AC-Ctrl software
onewire and dallastemperature are included only when TEMP_MEASURE_PERIOD >0. And if a library is missing then you get an error message during compile.

However, congratulation 👍

Aesculapius, 2022-04-14:
then I can't explain it :)

Aesculapius, 2022-04-15:
Okay, you were right I think.... it worked for a bit but now it doesn't anymore... and the values are way off again;

[image]

I'll remeasure the X1 voltages without PCB.

Aesculapius, 2022-04-15:
Without PCB connected:

1 - GND : 0V
2 - not measured because of risk of shorting with ground in this unhandy position ;-)
3 - MOSI : 4.66 V
4 - SCK : 4.7 V
5 - +12V : 12.46 V

absalom-muc, 2022-04-16:
I also think that it is related to the voltage shifter. But since it had worked for a short time, it could be related to a bad solder conncetion. I suggest you resolder all voltage shifter related solder connections (especially the voltage supply / GND pins). But please reduce the temperature to <400°C. I usually use approx. 350°C. And use some solder, just heating it probably won't do the trick.

Aesculapius, 2022-04-18:
I checked every connection to the levelshifter and verified the working of it. My first conclusion was wrong, as the component seems to need the 3.3V also on LV to work correctly (delivered by the Wemos). I was wrongly assuming it would just transform the 5V input to 3.3V itself...

I also reflashed everything from a clean project to make sure no unwanted typos were present (with two small kids around, you never know if they hit the keyboard when you weren't looking for 0.2 seconds lol).

I know get a different output, it seems. I looks like there is some short between fMISO and fMOSI, but then the multimeter says there isn't...

[image]

Also when sending a command to the set topic, I don't get an updated cmd_receiver=ok. So is it not reading the MQTT topic correctly or is it unable to send the command to the AC?

Aesculapius, 2022-04-18:
I'm lost....fSCK is back to <3000 now, fluctuating like mad...
[image]

absalom-muc, 2022-04-18:
Since fMISO=fMOSI I assume a short, too. If a MISO frequency>10Hz is detected during boot, then the ESP8266 stops working (endless loop until device is reseted). Which SW do you use? I prefer to do the debugging based on my SW only, nothing else should interfere.

Aesculapius, 2022-04-18:
SW you mean version (your src folder), compiled and uploaded with Arduino IDE. Last time I used Termite to look at the console but would happily use something different.

absalom-muc, 2022-04-19:
My request is only related to the ESP8266 SW in the src folder.
If resolder doesn't help I have no other idea than replacing the level shifter and/or the ESP8266

Aesculapius, 2022-04-20:
Well, since the soldering connections were so visibly right and clean, I used a razor knife to clean any dust or residue between pins. And I reflashed the Wemos with higher bitrate (but don't see that having anything to do with it). And I used an extension cable to prevent wifi drop-out. It has been running stable now for two days so I assume I did something right. 

Bit unsatisfying that I couldn't find any conclusive solution, but well... it doesn't seem to relate to your software so closing this issue.

closed 2022-04-20 by Aesculapius