absalom-muc/MHI-AC-Ctrl #85: No CNS connector soldered

kARTechnology, 2022-04-22:
I have 2 MHI ac,s 
one is variable inverter technology  (SRK-18YXS2-W6, 1.65 ton, 5.7* ISEER)
and another is non inverter fixed speed unit (SRK-25CWX-S6 - 1.95 ton, 4.2* ISEER) , **NO OUTDOOR UNIT COMMNUNICATION**

 I was able to successfully implement mhi-ac-ctrl in the inverter model and working fine since many months,
Today i opened the non-inverter model to add the CNS cable and check if mhi-ac-ctrl will work.. but to my surprise i found the connector is not soldered..

My question is how to know
[image]

[image]

 if the Indoor unit supports CNS monitoring? (Basic on/off/mode/temperature/speed/vales/swing)
before i solder on a CNS connector i want to know how i can test if the unit is outputting any data for communication?

absalom-muc, 2022-04-22:
It is also suprising for me. I don't know if the second unit provides the required SPI signals. You could measure the voltage on the pins, the levels described [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-pins-not-properly-connected). But this is complicated and dangerous becuase there are no solder pads on the top of the PCB.

kARTechnology, 2022-04-22:
> It is also suprising for me. I don't know if the second unit provides the required SPI signals. You could measure the voltage on the pins, the levels described [...]

It does look to have the pads, but no socket soldered. this is a single sided pcb only right?
it would not have the sensor values other than indoor temp sensor, pipe sensor.
in such case what would mhi-ac-ctrl try to collect from the unit? outdoor unit values are not available.

absalom-muc, 2022-04-23:
> It does look to have the pads, but no socket soldered. this is a single sided pcb only right? it would not have the sensor values other than indoor temp sensor, pipe sensor. in such case what would mhi-ac-ctrl try to collect from the unit? outdoor unit values are not available.

Solder pads seem to be on the bottom side only, so yes, it is a single sided pcb. I don't know which values are used from this indoor unit.

kARTechnology, 2022-05-11:
It seems like only the models with ODU communication cable support CNS. I found the backside of **pcb not having required components** for cns to work. but 12v and gnd available. need to see if i can implement IR control with a buck converter+esp8266
[image]

closed 2022-05-11 by kARTechnology