absalom-muc/MHI-AC-Ctrl #87: Read values in code

kARTechnology, 2022-05-16:
How to read the AC Power,Temperature,Fan,Swing status in arduino code itself? I was able to get mhi-ac-ctrl to send data to influ and also host a webserver to accept commands, but how do i read the values to display the current state...
`
 String("<html><body> <h1>MHI-AC-CONTROL "+String(status_power)+"-"+String(status_tsetpoint)+"</body></html>"));`
[image]

absalom-muc, 2022-05-16:
I suggest you use CallbackInterface_Status (row 176 in MHI-AC-Ctrl.ino). All status information are delivered to this class and inside the switch comand you can select the information you are interested. You could replace the output_p call by your comand to display the status.

kARTechnology, 2022-05-17:
> I suggest you use CallbackInterface_Status (row 176 in MHI-AC-Ctrl.ino). All status information are delivered to this class and inside the switch comand you can select the information you are interested. You could replace the output_p call by your comand to display the status.

Pls help me with example to read fanspeed/set temp/returnair..

absalom-muc, 2022-05-17:
Not sure related to your yellow and red marking. But e.g. below the line with power=1 or power=0 you could make the output to your display,

kARTechnology, 2022-05-19:
Thanks. Added web server and http updater, influxdb. and can use cron to send schedule by http commands(my mqtt broker is openwrt router)

[image]

closed 2022-05-19 by kARTechnology