absalom-muc/MHI-AC-Ctrl #88: Relevance of C1 and C2

WallnerK, 2022-05-17:
Hi,
thanks for this awesome project! I have been tinkering with it for some time due to some differences in the components I use. After I have blown out a capacitor (my fault haha) I recognized that the TSR-1-2450 does come with included capacitors to clean up the input voltage. That raised the question if C1 and C2 are technically even necessary?

I'm sure it is the right way to go, but I am just curious about the proper argument behind it. Thanks!

absalom-muc, 2022-05-17:
Originally I had a cable between MHI-AC-Ctrl and the AC, therefore my thinking was that the C's are beneficial. But since the output voltage from the CNS should be well regulated it is most probably not needed. Where have you find the information that TSR-1-2450 does come with included capacitors to clean up the input voltage?

WallnerK, 2022-05-17:
I got that info from the datasheet: https://cdn-reichelt.de/documents/datenblatt/D400/TSR1_D.pdf. On the first page it states that there are built in filtering capacitors.

There is a application note for input filtering, but that is only beneficial/required for voltages above 32VDC, no?

It just happens that I have been waiting for my order of ELKOs for a couple of weeks now... So thats why I investigated weather I could have my test setup without those for now.

absalom-muc, 2022-05-17:
yes, no problem to test without the ELKO. Thanks for the link.

DYLaKo, 2022-06-20:
Is it also possible to use another value for C1? In your design, there is 22uF, 25V, but the board voltage does not exceed 12V, right? So 22uF, 16V should work also?

absalom-muc, 2022-06-22:
yes, 22uF, 16V should work also. 25V is for increased robustness.

absalom-muc, 2022-08-05:
I believe the issue was addressed, therefore closed. If there are further questions, please feel free to re-open it.

closed 2022-08-05 by absalom-muc