absalom-muc/MHI-AC-Ctrl #90: SRF35ZS-V5ZS-W

mkaleu, 2022-05-23:
Hello, I bought a system with a floor type device and now I want to control it via WiFi. What do I have to do to get the necessary information.
Regards Michael

absalom-muc, 2022-05-23:
If your device is not listed [here](https://github.com/absalom-muc/MHI-AC-Ctrl#prerequisites) you could checkif it has a [CNS connector](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md#connector). If it has a CNS connector then the probability is high that it works, but finally you have to try it.

mkaleu, 2022-05-24:
It's not in the list but has a CNS connection. Should I now run the two test programs and send the results?

TimCinel, 2022-05-24:
Not quite the same series, but I can confirm that SRF50ZMXA-S (5kW floor console) has CNS and is compatible with the hardware and software in this repo.

Very tight fit, though:
[image]
.

kARTechnology, 2022-05-25:
> Thanks Tim, that's exactly what my console looks like. I've tried it, but somehow it didn't work. It may be due to the weak wifi. When I have my repeater I will try again. Reards Michael

as long as it has a cns connector it should work. at least get some replies from the unit that communication is happening. 
the ones without cns support will not have cns connector soldered in my experience and lack essential components required for cns communication.

mkaleu, 2022-05-28:
Hello friends, I now have a clean WiFi connection. I'm attaching the serial output because I don't think everything is going right yet. The ESP keeps reconnecting to the WiFi and nothing more than "connected = 0" arrives at the MQTT
Regards Michael
[Klima an heizen.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/8791067/Klima.an.heizen.txt)
.

absalom-muc, 2022-05-29:
Have you considered the [troubleshooting guide](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md)? Especially the reasons for [periodically crahses](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-esp8266-crashes-periodically)?

mkaleu, 2022-05-29:
Hello, as is probably always the case, the reference to PubSubClient v2.8.0 is skipped. I've changed it now and it works. Thank you. I will now continue to work on it.
[MQTT-Output.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/8793603/MQTT-Output.txt)

mkaleu, 2022-05-29:
Now I'm stuck again because I don't understand the syntax of the commands. What do I have to publish as an example if I want to switch the device on or off. The topic starts with MHI-Hueche/

kARTechnology, 2022-05-29:
Use MHI-Hueche/set/Tsetpoint with value as temperature..

MHI-Hueche/set/Fan with fan speed 1 to 4..

MHI-Hueche/set/Power with On/Off

mkaleu, 2022-05-29:
Thanks, it's working fine now.

closed 2022-05-29 by mkaleu