absalom-muc/MHI-AC-Ctrl #92: use 0.5 degrees as temp setting?

rwijnhov, 2022-06-07:
Is it possible to set the temp on 0.5 degrees somehow? I would  like to use a temp setting of 22.5 degrees if that is possibe somehow?

kARTechnology, 2022-06-07:
yes possible with latest firmware. but in my case if u turn off the ac and back on it loses the precision. dont know why.

absalom-muc, 2022-06-07:
Tsetpoint supports a resolution of 0.5°C, just added this information to the documentation.

>back on it loses the precision. dont know why

I can't confirm this behaviour for my AC, the resolution of 0.5°C is still given after a power off/on cycle.

rwijnhov, 2022-06-08:
What firmware update?

rwijnhov, 2022-06-08:
I already use v2.5R21

absalom-muc, 2022-06-08:
Please provide a log file.

absalom-muc, 2022-08-05:
I believe the issue was addressed, therefore closed. If there are further questions, please feel free to re-open it.

closed 2022-08-05 by absalom-muc

kARTechnology, 2022-10-11:
> Ok if i am reading the documentation right I should be able to set to .5
> degrees on this firmware. But that isn't possible only full degrees. So is
> that a bug?

You can set 24,24.5,25 like that only in 0.5 degree precision.