absalom-muc/MHI-AC-Ctrl #93: I built with the level shifter reversed and smoked my wemos

kARTechnology, 2022-06-07:
Hi,  I have 3 MHI ac and successfully running 2 of them on this project.
unfortunately i built a new board for the 3rd ac yesterday, and to keep aesthetics i had reversed the level shifter board upside down(components facing down)  but forgot to change the wiring layout and connected to the AC. immediately, there was burnt smell within 2-3 seconds and i removed it from the ac immediately.(its a cassette ac so the CNS pin is fairly easy to access)

so the power to the wemos was correct but the power to the level shifter was inverted(on both hv and lv side, ground and power were swapped). so what caused the wemos to fry even though wemos was wired correctly?

absalom-muc, 2022-06-07:
Maybe you haven shorten an output pin of the Wemos. But not sure, and I don't plan to check it in detail. I don't think it's worth the effort.

kARTechnology, 2022-06-15:
Changed wemos to new and old level shifter still works

closed 2022-06-15 by kARTechnology