absalom-muc/MHI-AC-Ctrl #95: fatal error: ESP8266WiFi.h: No such file or directory

nmg196, 2022-06-18:
Hi!
I'm trying to compile this in Arduino IDE but I can't see how to add the ESP8266WiFi library.  I can add the others like OneWire OK, but nothing comes up in the library search tool for "ESP8266WiFi" so how are you supposed to add/install this?

I've found my board (and selected it) but can't compile because of this :(

I've also tried in Platform.io but I have exactly the same problem there.

[image]

absalom-muc, 2022-06-18:
If I remember correctly ESP8266WiFi is known once you configure the Arduino IDE in the right way, i.e. have you added
[image]
(I use the German version, in English it is the "Additional Boards Manager URLs") in the preferences.

nmg196, 2022-06-19:
I did already do that so I could add my board (a Wemos LOLIN S2)  so I already had the extra board manager URL there.

However I've just realised I can compile if I select a Wemos D2 mini but not if I select the Wemos Lolin S2 mini.

No idea why it doesn't work :(

nmg196, 2022-06-19:
I've also tried with Platform.IO and I have the same problem with this plaformio.ini:

[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = arduino

maxgerhardt, 2022-06-19:
Crossposted to https://community.platformio.org/t/stuck-fatal-error-esp8266wifi-h-no-such-file-or-directory-when-compiling/28400 with possible solution approach.

nmg196, 2022-06-20:
OK - I found that I had an older D1 Mini Pro so I will use that instead (worked first time!). I'm not sure I know enough to get the Wemos Lolin S2 code working.

closed 2022-06-20 by nmg196