absalom-muc/MHI-AC-Ctrl #97: SRK50ZS Fan Speed Bug?

stosoorok, 2022-06-25:
Used MHI-AC-Ctrl HW: Wemos D1 Mini
Used MHI-AC-Ctrl SW: Latest

Bug with Fan Speed. Fan speed is off sync. Low is 2, Mid is 3, Hi is 4, "Auto" is 5 (should be high). And Real "Auto" is 6. Check the video I made:
Speed from remote: https://www.youtube.com/watch?v=jrqNGGq2ZaE
Speed from MHI-AC-Ctrl: https://www.youtube.com/watch?v=NB77X7XsCV0

How about other users or is it just 50ZS bug?

absalom-muc, 2022-06-27:
I haven't had a chance to dive deeper into this topic yet. Just one question for now: do the numbers 1-6 in your question refer to the normal AC status or to the OpData "IU-FANSPEED"? If I remember correctly, the numbering is different.

stosoorok, 2022-06-27:
Okay, I try to clear things up bit. These numbers 1-6 refer to MHI-AC-Ctrl Home Assistant dashboard "iu fan speed" so OpData. If I select the speed from the original remote, then 1-5 are fan speeds and op-data 6 is "auto". There are no level 1 fan speed in OpData, 2 is low, 3 is mid, 4 is hi and 5 jumps to "auto". I manually renamed "auto" to "turbo" or very high. 

```
type: custom:simple-thermostat
entity: climate.mhi_ac_ctrl
header: false
layout:
  mode:
    names: true
    icons: true
    headings: false
sensors:
  - entity: sensor.lumi_lumi_weather_5afff302_temperature
    name: Room Temp
  - entity: sensor.lumi_lumi_weather_5afff302_humidity
    name: Humidity
step_size: 1
control:
  hvac:
    heat_cool: false
  fan:
    low:
      name: Low
      icon: hass:fan
    medium:
      name: Mid
      icon: hass:fan
    high:
      name: High
      icon: hass:fan
    auto:
      name: Turbo
      icon: hass:fan
```
[image]

KostyaSha, 2022-06-27:
sound like some MAX after HI is needed?

KostyaSha, 2022-06-28:
Is it possible to switch to auto with original wifi module?

KostyaSha, 2022-06-28:
https://www.mhi-mth.co.jp/en/products/pdf/rlf012a102b_english.pdf
In fact there is ultra low  mode
[image]

stosoorok, 2022-07-03:
Just discovered, in heating mode, fan speed in "Auto" (or very high/turbo, MHI-AC-Ctrl in HA) I have "IU Fan Speed" (OpData) 8.

absalom-muc, 2022-08-05:
I believe the issue was addressed, therefore closed. If there are further questions, please feel free to re-open it.

closed 2022-08-05 by absalom-muc