absalom-muc/MHI-AC-Ctrl #98: Connecting the MHI-AC-Ctrl board stops Wemos from starting up

Aesculapius, 2022-07-04:
Hi there again.

Strange problem with a new module. It doesn't connect to WiFi and thus the airco stops after two minutes or so (https://github.com/absalom-muc/MHI-AC-Ctrl/issues/75#issuecomment-996266574 ?).

I have resoldered a new board since the C3-resistor was blown on old one. So ordered new board and made sure to solder neatly with enough room between contacts etc. Now when powering up the Wemos while attached to serial console, I can see the Mhi-Ac-Ctrl program booting up and falling into -3 or -4 error loop, which is expected as I haven't yet connected to the ZSX.

When I plug it into the board though, the Wemos isn't booting up at all. It's not connecting to WiFi nor anything else. Only thing I see is the undescriptive first line which always appears:

[image]

This leads me to think there is some kind of fault in the connections or components of the newly soldered board, but how do I troubleshoot that? I've checked for shortages between pins (none) and if 3.3V and 5V are present (they are) while connected to the board. 

Any hints on how to troubleshoot this part?

absalom-muc, 2022-07-04:
Is an external temperature sensor DS18X20 attached? If TEMP_MEASURE_PERIOD is >0 please set it to 0 in support.h.

Aesculapius, 2022-07-04:
No it's not, I haven't touched that part. (Using v2.5R3)
[image]

absalom-muc, 2022-07-04:
can you provide the log when the PCB is not connected to the AC? Not sure if it helps but currently I have no better idea.

Aesculapius, 2022-07-04:
Well, there really isn't any. When I connect just the Wemos, I get logging and I see the program start up, choose WiFi network and falling in error loop because of not seeing anything on SCK pin etc. As expected, you might say.
When I plug the Wemos onto the MHI-AC-CTRL board, all that disappears. In both scenario's, nothing yet connected to the AC. I would expect no different behavior when the Wemos is attached to the PCB.

absalom-muc, 2022-07-04:
o.k. and are you sure - sorry for the question - that it is a WEMOS-D1-MINI with the expected pins?

Aesculapius, 2022-07-04:
Well, they're coming from China so there's always a risk, but have arrived in a batch of which multiple ones work as expected, also in combination with mhi-ac-ctrl. Pic (though can be fake as well of course):
[image]

absalom-muc, 2022-07-04:
can you test it with another WEMOS-D1-MINI? There are not too many options left, could be a fault of the WEMOS-D1-MINI or a fault of a part on the MHI-AC-Ctrl PCB

Aesculapius, 2022-07-05:
Yeah it also doesn't work with a freshly programmed other Wemos. Suggests something wrong with one of the components indeed. No clue as to how to troubleshoot that though (my multimeter doesn't measure capacitors for example).

closed 2022-07-07 by Aesculapius