absalom-muc/MHI-AC-Ctrl #100: Capture Remote control sensor temperature

bradpeczka, 2022-08-14:
I have a MHI KX1:1-series ducted unit (FDU140KXE6F/FDC140KXES6).

The unit was supplied with the RC-EX1 remote control, which has a inbuilt temperature sensor and the ability to switch the 'Room' temperature as used by the unit to that which is reported by the remote. This is useful as it provides a more indicative value of actual room temperature rather than that which is reported by the return air sensor.

Published documentation from MHI (https://static1.squarespace.com/static/582c20ee59cc685797327410/t/5e4a8c2483fe885c7765bc25/1581943946578/Beijer+Ref+Service+Support+Handbook+19+Web+2.pdf) suggests that this value should be able to be captured at OpData value 4, but I don't understand enough of the SPI protocol to test for its presence.

So - not a bug, and apologies for raising it as such, but is it possible to augment the code to capture this value?

absalom-muc, 2022-08-14:
An issue is not only for bugs, also for ideas, request enhancements etc. - so, no problem.

I don't understand the motivation for your request. You can't use RC-EX1 in parallel with MHI-AC-Ctrl, so reading back the value seems to be not appropriate.
If your motivation is in general using an external temperatures sensor instead the build-in room temperature sensor, you can do it with a DS18x20 temperature sensor soldered on the MHI-AC-Ctrl PCB or by providing the room temperature from an external sensor via MQTT, please have a look in the [configuration options for room temperature](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#room-temperature).

bradpeczka, 2022-08-16:
Looks like I have a bit more reading to do - thanks for you reply!

closed 2022-08-16 by bradpeczka