absalom-muc/MHI-AC-Ctrl #101: setpoint changes on its own

Aesculapius, 2022-08-16:
I've been using the modules with succes for the last weeks. One apparent issue is that if I keep the modules behind the plastic, they get a too weak WiFi signal somehow and get stuck into some kind of loop. So they hang out of the unit, but that's fine with me.

On the unit downstairs I've notices for the second time now that the setpoint is changed to 33 degrees. It has been set to 24 degrees for the whole day and nothing has changed. The remote is stored into a drawer thus also cannot have been interfering. As you can see in the MQTT details, the setpoint as sent to the module is 24 degrees, but apparently it changed by itself to 33?

Not sure if this can be a software bug or whether it's a MHI ac bug. This is on a SRK-50ZSX-W

[image]

[image]

Also not sure how to debug? It happend without apparent reason - was not fiddling around or adjusting any settings...

Aesculapius, 2022-08-16:
Now that I look at my temperature graphs, I see that it has occurred more often on this unit. No issue on the other unit though...

[image]

Aesculapius, 2022-08-17:
Yeah strange eh.
The AC indeed stops cooling as it feels no need to cool anymore (roomtemp is lower).
It is not automatically returned to the original value. At least, not the times that I noticed it. Needed to set the airco off and on again to have it reset the setpoint.
I've not seen it before 2.6R1. I'm happy to test with 2.5, but first want to see how reproduceable it actually is (hasn't occurred since posting this). Might try in a few days

JoepWi, 2022-09-17:
I have the same issue with one of my units (MHI SRF25ZS-W) if set to heating. It raises the temperature on it's own. eg. Set to 22 and after a few seconds it's changed to eg. 24.
[logs_mhi-kantoor-joep_logs_problem.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/9591419/logs_mhi-kantoor-joep_logs_problem.txt)
During the test above I had all HA automations disabled for this Airco unit.
The change of the target temperature is in my opinion coming from the MHI-AC-Ctrl unit.
Any help is welcome.

joshuammex, 2022-09-18:
@JoepWi I have the same issue as yourself with a SETSRK25ZSA-W, and have since the get go. AC randomly increases the setpoint by 2c shortly after I manually set the temp. I'm running the [MHI-AC-Ctrl-ESPHome](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome) fork.

absalom-muc, 2022-09-18:
I don't know the HA implementation. What is displayed in the row
`[12:06:16][D][climate:414]:   Target Temperature: 22.00°C`?
It is the standard Tsetpoint (listed in the table in this [clause](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status)) or is it the [opdata](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#operating-data-mhi-ac-ctrl-coreh) Tsetpoint?
It is known that the opdata Tsetpoint is increased by 2°C for heating when AC is powered on and decreased when AC is powered off. This is a normal behaviour of the AC and has nothing to do with MHI-AC-Ctrl.
In general it is preferred when you provide the log from MHI-AC-Ctrl, not from an addon using it.

JoepWi, 2022-09-18:
I traveling at the moment. Will look into it in a few weeks. Thanks for the explaination.

hjkoster400d, 2022-09-27:
> I don't know the HA implementation. What is displayed in the row `[12:06:16][D][climate:414]: Target Temperature: 22.00°C`? It is the standard Tsetpoint (listed in the table in this [clause](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status)) or is it the [opdata](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#operating-data-mhi-ac-ctrl-coreh) Tsetpoint? It is known that the opdata Tsetpoint is increased by 2°C for heating when AC is powered on and decreased when AC is powered off. This is a normal behaviour of the AC and has nothing to do with MHI-AC-Ctrl. In general it is preferred when you provide the log from MHI-AC-Ctrl, not from an addon using it.
> 
> edit: The opdata 2 Tsepoint MQTT topic should be "SET-TEMP", not "Tsetpoint". This will be fixed in future.

Hi, 

Is the AC actually targeting a 2 degrees higher temp then was set? Would it be a valid assumption that when the AC is in heat mode, the targettemp set by the user is 2 degrees lower the AC is reporting?

Thanks for now,

Gr Rik

absalom-muc, 2022-09-27:
> Hi,
> 
> Is the AC actually targeting a 2 degrees higher temp then was set? Would it be a valid assumption that when the AC is in heat mode, the targettemp set by the user is 2 degrees lower the AC is reporting?
> 
> Thanks for now,
> 
> Gr Rik

I don't think so, but I have no definite answer for you. IMO HA should use the standard Tsetpoint (listed in the table in this [clause](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status)) not the [opdata](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#operating-data-mhi-ac-ctrl-coreh) Tsetpoint. With the standard Tsetpoint there should be no increase of the value when heating.

ervee, 2022-09-29:
> @JoepWi I have the same issue as yourself with a SETSRK25ZSA-W, and have since the get go. AC randomly increases the setpoint by 2c shortly after I manually set the temp. I'm running the [MHI-AC-Ctrl-ESPHome](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome) fork.

I have the same with my split unit.  Also in the HA implementation. I did not see this during summer when cooling, but now with heating it likes to sometimes set the temperature +2 °C.

JoepWi, 2022-10-03:
Below 3 screenshots from the ESPHome interface (no HA involved!):

**Start screen Airco OFF**
[image]

**Airco manual on this webpage set to HEAT**
[image]

**When the fan of the indoor units starts blowing the temp increases 2 degrees**
[image]

I have this issue with all 4 indoor units I have.

JoepWi, 2022-10-06:
Nobody any idea?

ervee, 2022-10-06:
I just used my AC to heat. In HA it was off and set temp was 21C. I powered on the AC with the iR remote set to 21C. When I wanted to switch the AC off with HA I saw it set on 23C again but when I pressed the off button the set temp immediately went back to 21. So perhaps it is just cosmetic? And I see absalon-muc mention something about the Tsetpoint usage for the HA implementation being incorrect. Perhaps @ginkage can say something about this.
It could well be that the AC sets its internal setpoint to +2°C because of warm air rising to the top mounted sensor...

PS this perhaps has nothing to do with the original post. His temperature increases it a bit more than 2°C.

mikeschenkel, 2022-10-26:
This behavior has nothing to do with the MHI-AC-Ctrl software. An offset of 2°C in heating operation is determined by the AC itself. You can turn this off by holding down the ON/OFF button on the AC for 30 seconds while in heating operation. After you hear a beep and set a new temperature with the remote the offset should be resolved.

[image]

Aesculapius, 2022-10-27:
Haven't had troubles for a while but now, in the heating season, the issue seems to return. Now I'm seeing it on the tRoom value.

I'm setting the temp via mqtt every 20 seconds via /set/Troom:
[image]

but still the units own tRoom in /Troom keeps changing to other values now and then (plus 2,5 degree):
[image]

Seems to be a similar problem, not sure what to do next. Other thing is that at once I also have strange fSCK values (and thus the unit cannot be controlled also). Might be connected. I'll have to power-cycle everything to get it working again...
[image]

tomm1ed, 2022-10-28:
> Haven't had troubles for a while but now, in the heating season, the issue seems to return. Now I'm seeing it on the tRoom value.
>
> I'm setting the temp via mqtt every 20 seconds via /set/Troom: [...]

I am having similar issues:
https://github.com/absalom-muc/MHI-AC-Ctrl/issues/105
Maybe you can add to that issue? (since your Tsetpoint issue has been resolved maybe also close this issue?)

absalom-muc, 2022-10-28:
> This behavior has nothing to do with the MHI-AC-Ctrl software. An offset of 2°C in heating operation is determined by the AC itself. You can turn this off by holding down the ON/OFF button on the AC [...]

Thank you @mikeschenkel , I've seen a lot of different manuals for the AC, but never noticed that the 2°C offset was addressed.

Aesculapius, 2022-11-02:
> Haven't had troubles for a while but now, in the heating season, the issue seems to return. Now I'm seeing it on the tRoom value.
>
> I'm setting the temp via mqtt every 20 seconds via /set/Troom: [...]

As for my issue with strange SCK value (0), this was caused by a not infinite resistance between the pins of C2. Looked like some kind of corrosion. After cleaning that up and scraping off residue, everything worked again as intended. 

I also changed the ROOM_TEMP_MQTT_SET_TIMEOUT interval to 90 as per https://github.com/absalom-muc/MHI-AC-Ctrl/issues/105#issuecomment-1295017726 which seems to work so far...

tomm1ed, 2022-11-04:
> This behavior has nothing to do with the MHI-AC-Ctrl software. An offset of 2°C in heating operation is determined by the AC itself. You can turn this off by holding down the ON/OFF button on the AC [...]

Does this actually work for anyone? Tried this 3 times but no bibibi sound from my SRK 35 ZSX and the machine goes full blast when it turns on. The temperature is 18.9 at the external temp sensor which is translated to 18.5 on the MHI-AC-CTRL and at the lowest setting of 18 this means the unit tries to heat until it’s 20 thinking it has a delta of 1.5 degrees. Very loud and very high usage is the result so I would like to fix this by following this manual. can anyone explain exactly how to do this?

buennerbernd, 2022-11-04:
Have you tried eco-mode and silent-mode?
I would assume this saves some energy.

tomm1ed, 2022-11-04:
> Have you tried eco-mode and silent-mode? I would assume this saves some energy.

I have not and I don't intend to ;) I want my Airco to use the correct Setpoint and not Setpoint +2. Of this option does not work with the ZSX I will just feed it TRoom+1 or +2 but would prefer to use correct values.

tomm1ed, 2022-11-04:
> > This behavior has nothing to do with the MHI-AC-Ctrl software. An offset of 2°C in heating operation is determined by the AC itself. You can turn this off by holding down the ON/OFF button on the AC for 30 seconds while in heating operation. After you hear a beep and set a new temperature with the remote the offset should be resolved.
> > [image]
> 
> Does this actually work for anyone? Tried this 3 times but no bibibi sound from my SRK 35 ZSX and the machine goes full blast when it turns on. The temperature is 18.9 at the external temp sensor which is translated to 18.5 on the MHI-AC-CTRL and at the lowest setting of 18 this means the unit tries to heat until it’s 20 thinking it has a delta of 1.5 degrees. Very loud and very high usage is the result so I would like to fix this by following this manual. Could anyone please explain to me exactly how to do this? @mikeschenkel maybe ?
> 
> [image]

Too be 100% sure, they are talking about the On/Off button on the Remote yes? (@mikeschenkel )

mikeschenkel, 2022-11-04:
@tomm1ed No, you they're talking about the physical ON/OFF button on the AC unit itself.

tomm1ed, 2022-11-04:
> @tomm1ed No, you have to hold down the physical ON/OFF button on the indoor AC unit itself.

Thank you you are a life saver :) Held it for over 30 secs and bipbip it works 👍

buennerbernd, 2022-12-09:
I have tested this too. 
In eco mode the new behavior is: OpData/Tsetpoint = Tsetpoint - 1
Without eco mode: OpData/Tsetpoint = Tsetpoint

Aesculapius, 2023-01-19:
As per my comment on 2nd of November, this issue hasn't occurred and I feel it was related to both that setting as my local hassle with corrosion on the pcb. The posts after that discuss a different topic so maybe should be split and this one closed?

closed 2023-01-19 by Aesculapius