absalom-muc/MHI-AC-Ctrl #102: TSR 1-2450e in pcb design

hjkoster400d, 2022-09-03:
Hi there,

First of all, thanks for sharing this awesome project.
I was wondering if you considered using the tsr 1-2450e instead of the non-e variant. The e variant has built-in capacitors so it removes the need for the 2 external ones. 

Any thoughts?

Greets Rik

absalom-muc, 2022-09-03:
Thank you for pointing me to TSR1-2450e.
Have you found a comparison TSR1-2450 vs. TSR1-2450e? Spontaneously I found only that TSR1-2450e has a slightly lower Voltage Set Accuracy of +/-4% compared to TSR1-2450 of +/-2% (which should be no issue for our application). For both an internal filter with internal capacitor is mentioned.

In general, I don't know if the planned 22µ and the 100n capacitor are needed. According to the datasheet of TSR1-2450 for input voltage higher than 32 VDC (we have 12V) the external capacitor is needed and the 100n capacitor is not mentioned. However, I added them from the beginning to reduce noise. I could imagine that they are unnecessary. But to be honest, I was not motivated to make exhaustive tests without the external capacitors.

hjkoster400d, 2022-09-03:
The e-variant has a bit less efficiëncy (92% instead of 94%) also the noise reduction is a bit better when using the 1-2450. But the price of the e-variant is lower, and it saves the 'need' for the external capacitors. I've some additional pcb's laying around, i will try to put a 2450e and let you know the results.

absalom-muc, 2023-01-02:
@FloR012 confirmed that it works with the 'E'-version.

closed 2023-01-02 by absalom-muc