absalom-muc/MHI-AC-Ctrl #103: Info on accessing the main board?

jeroen94704, 2022-09-14:
I made a couple of these modules (and have a bunch of spare unpopulated PCB's if anyone is interested), but for some reason I blindly assumed the installation would be as simple as lifting the front cover of the unit and removing the black cover plate behind it. Unfortunately is seems like it's a bit more involved than that, and while I'm sure I can figure it out it would greatly put my mind at ease if I could see some pictures or a video about the process. Is that available somewhere? I couldn't find anything on YT. 

The specific models I have are the SRK35 ZS-W and SRK50 ZS-W.

fonske, 2022-09-14:
A YT video (assuming you speak dutch)
https://www.youtube.com/watch?v=b_oNoAsSqIE

You only have to do it once if you are using the silicone cable (20 cm) to put it to front of the unit and not directly connect it to the motherboard.
See:
https://tweakers.net/fotoalbum/image/ismYXEzDBsUAXDobtcaeQGuY.jpg
https://tweakers.net/fotoalbum/image/RYFaBsBlIiSJVQtQOlmCcRRK.jpg
https://tweakers.net/fotoalbum/image/jx4zreOrfBud9odKrivhV774.jpg

But you need to change the posistion of the connector on the print.

jeroen94704, 2022-09-14:
Thanks! I did see that video, but as far as I saw they don't show how you get to the actual control board, right? Did you happen to snap any pictures of that?

jeroen94704, 2022-09-14:
Awesome, thanks!