absalom-muc/MHI-AC-Ctrl #105: Troom spikes when using MQTT as temperature source

tomm1ed, 2022-10-24:
I am using an external Temperature sensor to set the Troom value on my SRC35ZSX but I see weird spikes when heating in the value like the internal sensor also keeps updating (near the Airco of course it is hotter than at the temp sensor which is on the other end of the room)

See https://tweakers.net/fotoalbum/image/ax3hBggwaNhKPqMj9ZOg04s7.jpg for the temperature the Airco is reporting in the MQTT topic and https://tweakers.net/fotoalbum/image/CPcGYRJTto0jOrhyxENbRJuf.jpg for the actual temperature measurement.
The value of Troom is updated every minute from the ESP (running Tasmota)

Blokkert, 2022-10-24:
Same issue here.

see: https://tweakers.net/fotoalbum/image/W3rlt7D76CKyctGQlkOCe448.png
both values in one graph.

its worst on the moment the unit is in heating mode, spikes continously.
Tried different things in the mqtt.publish, but problems stays everytime.

Current code:

```
 # Airco

- id: Tuya Room Temp To AC-Living
  alias: Tuya Room Temp To AC-Living
  trigger:
  - platform: time_pattern
    minutes: /1
#  - platform: state
#    entity_id:
#      - sensor.temptuya1_temperature
  condition:
  - condition: template
    value_template: "{{ states('sensor.temptuya1_temperature')|is_number }}"
  action:
  - service: mqtt.publish
    data:
      topic: AC_MHI_Living/Troom
      payload_template: "{{ states('sensor.temptuya1_temperature') }}"
  mode: single
```

tried this to, no difference.

`payload_template: '{{ states.sensor.temptuya1_temperature.state }}'`

tomm1ed, 2022-10-24:
My setup uses Tasmota with the following rules:
```
rule1 ON BMP280#Temperature DO Var1 %value% ENDON
rule2 ON Time#Minute|1 DO publish MHI-AC-Ctrl-Livingroom/Troom %Var1% endon
rule1 1
rule2 1
```

tomm1ed, 2022-10-24:
Created a graph with both values superimposed (never knew this was possible, thanks @Blokkert ;) )
 
[image]

absalom-muc, 2022-10-24:
Can you provide a log file? Maybe something happens in between.

tomm1ed, 2022-10-24:
> Can you provide a log file? Maybe something happens in between.

A log file of what exactly?

absalom-muc, 2022-10-24:
The log of the serial interface to see what happens between 2 spikes.

tomm1ed, 2022-10-25:
> The log of the serial interface to see what happens between 2 spikes.

Okay, excuse for being a bit daft but how can I access this log?

Blokkert, 2022-10-25:
i did check something reqarding mqtt timeout, its configured on 20 sec. So i changed the mqtt.publish to every 10 seconds, issue stays te same.

absalom-muc, 2022-10-25:
> > The log of the serial interface to see what happens between 2 spikes.
> 
> Okay, excuse me for being a bit daft but how can I access this log?

If you used the Arduino IDE for programming the MHI-AC-Ctrl board, you can use the serial monitor of the Arduino IDE for displaying the log output. Please copy the log to a text file.

absalom-muc, 2022-10-25:
> i did check something reqarding mqtt timeout, its configured on 20 sec. So i changed the mqtt.publish to every 10 seconds, issue stays te same.

When you mention the mqtt.publish, I assume you mean the publish from your external temperature sensor - correct?

PietNL, 2022-10-26:
As I'm also using this feature (also with HA) and not seeing these spikes, I just compared my automation with yours and noticed that you are using this:

```action:
  - service: mqtt.publish
    data:
      topic: AC_MHI_Living/Troom
      payload_template: "{{ states('sensor.temptuya1_temperature') }}"
```

And I'm posting the temp sensor to a different topic(including the set):

```
action:
  - service: mqtt.publish
    data:
      topic: AC-Living/set/Troom
      payload_template: "{{ states('sensor.living_temperature') }}"
```

If you try this, does that make a difference?

tomm1ed, 2022-10-26:
> As I'm also using this feature (also with HA) and not seeing these spikes, I just compared my automation with yours and noticed that you are using this:
> 
> ```
>   - service: mqtt.publish
>     data:
>       topic: AC_MHI_Living/Troom
>       payload_template: "{{ states('sensor.temptuya1_temperature') }}"
> ```
> 
> And I'm posting the temp sensor to a different topic(including the set):
> 
> ```
> action:
>   - service: mqtt.publish
>     data:
>       topic: AC-Living/set/Troom
>       payload_template: "{{ states('sensor.living_temperature') }}"
> ```
> 
> If you try this, does that make a difference?

Do you mean you don’t see these spikes on your install?

tomm1ed, 2022-10-26:
> If you used the Arduino IDE for programming the MHI-AC-Ctrl board, you can use the serial monitor of the Arduino IDE for displaying the log output. Please copy the log to a text file.

Ah thank you, now I feel double daft. As my airco is quite high reaching the USB port is very hard, so I am hoping someone else can supply this info. (Or is there any way to get this info over the network?)

JoepWi, 2022-10-26:
> Do you mean you don’t see these spikes on your install?

The difference is just a naming of the entities in your installation and configuration. That has nothong to do with the spikes.

tomm1ed, 2022-10-26:
> The difference is just a naming of the entities in your installation and configuration. That has nothong to do with the spikes.

Thank you for you reply! My question is in my view quite a relevant question. Does the Troom follow your ‘set/Troom’ command and does it also result in no spikes?

Also; where did you get this info? 
I am running the version before the most current one and this topic was not mentioned anywhere as far as I remember. I am also unable to find the ‘set/Troom’ in the current documentation nor MHI-AC-CTRL.h so I am wondering how you knew to use that topic.

absalom-muc, 2022-10-26:
Great finding @PietNL , thank you!

For MQTT publish to MHI-AC-Ctrl you have to use the set-path, please compare with [SW-Configuration.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status)
Publishing to AC_MHI_Living/Troom has no effect (not considered by MHI-AC-Ctrl) and will be overwritten by MHI-AC-Ctrl.

Blokkert, 2022-10-26:
> Great finding @PietNL , thank you!
>
> For MQTT publish to MHI-AC-Ctrl you have to use the set-path, please compare with [...]

Hmm, i already tried the /set/ topic earlier without any results, the troom value goes directy to internal temp sensor. So does'nt work at all here..

tomm1ed, 2022-10-26:
On my MQTT broker this is what I see when using /set/Troom topic:
'''
MHI-AC-Ctrl-Livingroom/set/Troom 18.6
MHI-AC-Ctrl-Livingroom/cmd_received unknown command
'''

Blokkert, 2022-10-26:
my version is set on 2.5R21 also, i can upgrade this to 2.6R2 and will report back when completed. Doing this somewhere this week.

tomm1ed, 2022-10-26:
Okay, just checked my sourcecode triggered by:

and lo and behold...
//#define ROOM_TEMP_MQTT                            // use room temperature from received MQTT topic

tomm1ed, 2022-10-27:
It appears I spoke too soon :( 
It's gotten even worse:

In MQTT I see:
MHI-AC-Ctrl-Livingroom/set/Troom 19.2
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/Troom 19.00
MHI-AC-Ctrl-Livingroom/Troom 23.00
MHI-AC-Ctrl-Livingroom/set/Troom 19.2
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/Troom 19.00

Blokkert, 2022-10-27:
Seems that the problems are solved here.

1. I upgraded to 2.6R2
2. mqtt timeout is set on 60 seconds
3. added /set/ in the payload
4. my mqtt.publish from HA is running every 45 seconds

Spikes are gone.

I noticed, the troom has a accuracy of 0.25 degrees.
 Sending value of 19,67 results of troom of 19,5
19,81 results in 19,75.

But thats all fine.

You can see the moment i fixed everything according above listed points:

[image]

tomm1ed, 2022-10-27:
> 

Looks like I need to migrate to 2.6R2 then... I will do this tomorrow.

tomm1ed, 2022-10-28:
Updated to 2.6R2 and it looks like it's better, but not fixed. The spike to 23 returns now and then:
```
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/Troom 19.50
MHI-AC-Ctrl-Livingroom/OpData/THI-R3 17
MHI-AC-Ctrl-Livingroom/OpData/CT 0.00
MHI-AC-Ctrl-Livingroom/OpData/OU-FANSPEED 0
MHI-AC-Ctrl-Livingroom/OpData/DEFROST Off
MHI-AC-Ctrl-Livingroom/OpData/TOTAL-COMP-RUN 200
MHI-AC-Ctrl-Livingroom/OpData/OU-EEV1 250
MHI-AC-Ctrl-Livingroom/OpData/RETURN-AIR 19.50
MHI-AC-Ctrl-Livingroom/OpData/TDSH 0
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/OpData/TDSH 0
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/OpData/THI-R2 64
MHI-AC-Ctrl-Livingroom/OpData/OUTDOOR 12.00
MHI-AC-Ctrl-Livingroom/OpData/THO-R1 3
MHI-AC-Ctrl-Livingroom/OpData/THO-R1 2
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/OpData/THO-R1 3
MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/OpData/THO-R1 2

MHI-AC-Ctrl-Livingroom/Troom 23.00

MHI-AC-Ctrl-Livingroom/set/Troom 19.5
MHI-AC-Ctrl-Livingroom/cmd_received o.k.
MHI-AC-Ctrl-Livingroom/Troom 19.50
[...]
```
Also visible in the Graphs in Home Assistant:
https://tweakers.net/fotoalbum/image/snApcg4r78KqcVZVV0EqAEfO.png
Like I said, my other ZSX with external DS18x20 does not have this issue. @absalom-muc any insight would be much appreciated

tomm1ed, 2022-10-28:
So that indeed seens to fix it. 
If  the update by Tasmota took even a microsecond longer than 60 seconds MHI-AC-CTRL defaulted back to the internal temperature and was then quickly updated again by Tasmota. 
Since changing the ROOM_TEMP_MQTT_SET_TIMEOUT to 90 no spikes at all:
ROOM_TEMP_MQTT_SET_TIMEOUT

mjszwo, 2022-11-10:
I have the exact same issue with two Wemos D1. In my case, it does not matter if I let the aircon decide the room temperature, or if I push/force the temperature by MQTT.

Strange enough the spikes occur at TOPIC_TROOM, whilst TOPIC_RETURNAIR doesn't show the spikes at all. So I compared what is different between the two topics and I found this:

```
       case opdata_return_air:
        **case erropdata_return_air:**
          dtostrf((value - 61) / 4.0, 0, 2, strtmp);
          output_P(status, PSTR(TOPIC_RETURNAIR), strtmp);
          break;
		  
		  
        case status_troom:
          dtostrf((value - 61) / 4.0, 0, 2, strtmp);
          output_P(status, PSTR(TOPIC_TROOM), strtmp);
          break;
```

Is it possible that the spikes at TROOM will be resolved if 'case erropdata_troom:' is added, or am I thinking too simple?

absalom-muc, 2022-11-12:
@mjszwo : Please provide a log of the serial debug output

mjszwo, 2022-11-13:
> @mjszwo : Please provide a log of the serial debug output

Meanwhile I added my suggestion myself and I had no spikes for ~48 hours. This morning I flashed my Wemos again without the change and started logging. Here's the output where two spiked occured:

```
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
status=136 topic=THI-R2 payload=56
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
status=143 topic=TD payload=<=30
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
mhi_ac_ctrl_core.loop error: -4
status=136 topic=THI-R2 payload=56
mhi_ac_ctrl_core.loop error: -4
status=143 topic=TD payload=<=30
mhi_ac_ctrl_core.loop error: -4
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Mode payload=Fan payload_length=3
status=64 topic=cmd_received payload=o.k.
status=65 topic=Mode payload=Fan
status=64 topic=Power payload=On
status=65 topic=Mode payload=Fan
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
mhi_ac_ctrl_core.loop error: -4
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Mode payload=Heat payload_length=4
status=64 topic=cmd_received payload=o.k.
status=65 topic=Mode payload=Heat
status=148 topic=unknown payload=4306
Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
status=130 topic=Tsetpoint payload=22.0
status=147 topic=OU-EEV1 payload=200
mhi_ac_ctrl_core.loop error: -4
status=148 topic=unknown payload=4306
Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
status=148 topic=unknown payload=4306
Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
status=141 topic=COMP payload=10.00
status=135 topic=THI-R1 payload=16
status=148 topic=unknown payload=4306
Unknown operating data, MOSI_frame[DB9]=210 MOSI_frame[D10]=16
status=136 topic=THI-R2 payload=55
status=138 topic=OU-FANSPEED payload=5
status=147 topic=OU-EEV1 payload=150
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
status=135 topic=THI-R1 payload=19
status=136 topic=THI-R2 payload=53
status=134 topic=IU-FANSPEED payload=1
status=132 topic=OUTDOOR payload=8.50
status=133 topic=THO-R1 payload=-4
status=68 topic=Troom payload=29.25
status=68 topic=Troom payload=21.25
status=141 topic=COMP payload=59.70
status=143 topic=TD payload=<=30
status=137 topic=THI-R3 payload=23
status=142 topic=CT payload=2.20
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
status=144 topic=TDSH payload=0
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
status=147 topic=OU-EEV1 payload=350
mhi_ac_ctrl_core.loop error: -4
status=135 topic=THI-R1 payload=26
status=136 topic=THI-R2 payload=45
status=134 topic=IU-FANSPEED payload=8
status=132 topic=OUTDOOR payload=8.75
status=133 topic=THO-R1 payload=-2
status=141 topic=COMP payload=44.70
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
status=143 topic=TD payload=<=30
status=137 topic=THI-R3 payload=26
status=142 topic=CT payload=1.65
status=144 topic=TDSH payload=1
mhi_ac_ctrl_core.loop error: -4
status=147 topic=OU-EEV1 payload=317
mhi_ac_ctrl_core.loop error: -4
status=135 topic=THI-R1 payload=25
status=136 topic=THI-R2 payload=51
MQTT_subscribe_callback, topic=MHI-AC-Ctrl-US/us/set/Troom payload=21.25 payload_length=5
ROOM_TEMP_MQTT: 21.250000 146 146
status=64 topic=cmd_received payload=o.k.
status=132 topic=OUTDOOR payload=9.00
status=133 topic=THO-R1 payload=-1
```

And here a screenshot of the spike. I suspect the first spike to be the non-MQTT temperature because of the reboot. But the second spike is the one that occurs randomly.

[image]

tomm1ed, 2022-11-13:
@absalom-muc / @mjszwo for what it’s worth since upgrading to 2.6R2 with the 90 second timeout I have had zero spikes:

[image]

absalom-muc, 2022-11-13:
I can't imagine that your problem is related to 'case erropdata_troom'. As far as I know erropdata_troom doesn't exist.

If you used so far 80MHz for CPU frequency, you could test if the number of  "mhi_ac_ctrl_core.loop error: -4"-messages is reduced by applying 160MHz.

If it doesn't help please provide the log file in the way I described [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#recording-a-basic-log-file) (added few hours ago), especially the time stamp should be enabled. Addtionally please provide the following information:
- SW version
- ROOM_TEMP_MQTT_SET_TIMEOUT value (but the analysis could be simplified when you don't use writing troom via MQTT)

mjszwo, 2022-11-16:
I have been running the Wemos on 160MHz for +48 hours, and no spikes occured within that period. So it seems that this fixed it.

closed 2022-11-17 by absalom-muc