absalom-muc/MHI-AC-Ctrl #108: Question: Disable fan during heating pauses

buennerbernd, 2022-10-31:
I have a problem when I‘m using my aircon for heating. The heating works well until the temperature is achieved. The heating is off but the fan is still blowing. This feels uncomfortable. I would prefer the fan was also off during the heating pauses. 
Has anyone observed this as well?
Has someone an idea to overcome this problem by using MHI-AC-Crtl and some smart rules?

Thank you for input.

mikeschenkel, 2022-10-31:
I'm not sure if this is something that MHI-AC-Ctrl can solve. You could try selecting different settings via the remote to find the behavior that suits your needs best in heating operation.

[image]

buennerbernd, 2022-11-01:
Thank you Mike, this was exactly I was looking for. Unfortunately non of my 3 devices provides this menu option.
I have SRK35ZS-S, SRK20ZS-S and SRF35ZMX-S.
The SRKxxZS-S devices only provide L2 and CL in the menu.
The SRF35ZMX-S device even don't have a menu, just a "clean" button on this place.

I would love to test the options F1 and F2 as you found in the manual. What kind of model is this?
Now I'm thinking about simulating this function without destroying the devices.
Has some experiences with that?

mikeschenkel, 2022-11-04:
@buennerbernd My AC units are SRKxxZS-W.

buennerbernd, 2022-11-08:
Sounds like the SRKxxZS-W is a minor update of SRKxxZS-S.
I found out that wired remote controls like RC-EX3A also support this feature "Fan control in heating thermo-OFF".
But I assume these controls are not wired over the CNS port.
I would be interesting if the SRKxxZS-S also accepts this setting.

buennerbernd, 2022-11-29:
I have implemented this missing feature that way:
When heating:
If the /OpData/IU-FANSPEED goes down to 1: /set/Power Off.
Wait until /OpData/RETURN-AIR - 1 < /OpData/Tsetpoint: /set/Power On.

Despite the off-intervals are longer I don't have that cold draft.

closed 2022-11-29 by buennerbernd

absalom-muc, 2022-11-29:
Hi @buennerbernd ,
not sure if this is a good solution because you could reduce the lifetime of your AC by the increased numbers of on/off cycles. 
Just my 2 cents

buennerbernd, 2022-11-29:
Of course I'm also not absolutely sure if this could reduce the lifetime.
I have observed and logged the system for some days before I started programming. 
For my system  /OpData/IU-FANSPEED = 1 is only at start-up or in a heating pause. In the latter case the outdoor unit is always shut down.
So my assumption was that these pauses are very similar to on/off cycles.
As I mentioned, the off-intervals are now longer than before, because the internal sensor cools down slower than before.
The on-phase is also longer than before. So in summery I have far less cycles than before.

Of course I'm interested if someone has any objections.

absalom-muc, 2022-11-30:
o.k., that sounds reasonable.