absalom-muc/MHI-AC-Ctrl #110: Floor model with 2 vanes to control?

Dust-E, 2022-11-04:
Hi all,

I have a MHI SRF50 ZSX-W and this (floor model) indoor unit has 2 vanes to control. Above and below.
Is it possible to operate both?

[image]

Thank you in advance

absalom-muc, 2022-11-05:
No, it is not supported.

Dust-E, 2022-11-05:
Thanks for replay.
Is this something that can be made and if so, do you want to make it?

absalom-muc, 2022-11-05:
I don't know if it is possible, the following topics would have been clarified:
1. Is SRF50 ZSX-W in general supported by MHI-AC-Ctrl?
2. Is it visible on the SPI bus when the 2 vanes are controlled via the IR remote control? (even for other ACs with only 1 vanes it works not as expected)
3. Someone with this AC model would have to record the SPI communication and identify the according SPI commands.
4. With this knowledge it could be easy added to MHI-AC-Ctrl.

To be honest, I don't want to spend the time supporting clarifying the topics above, especially topics 2 and 3 - sorry.

damienmorrissey, 2022-11-05:
I have two of these units. I can't recall if the remote has the control for
it of it it's via the panel only. I might be able to get a dump of stuff.
Is there a guide on how to do it?

absalom-muc, 2022-11-06:
Please check [MHI-AC-Trace](https://github.com/absalom-muc/MHI-AC-Trace) and some chat about it in [issue 77](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/77)

Dust-E, 2022-11-06:
I can control the AC with the MHI-AC-Ctrl via HA
You can only operate the bottom vane via the control on the AC and not on the remote

glsf91, 2023-10-02:
Closed because of no activity for a long time

closed 2023-10-02 by glsf91