absalom-muc/MHI-AC-Ctrl #111: Wrong assembly orientation of X1 in KiCad PCB layout

absalom-muc, 2022-11-05:
@oliv3r : Not sure if you have seen my question in your [KiCad pull request](https://github.com/absalom-muc/MHI-AC-Ctrl/pull/106).

[image]

 I believe the orientation of X1 is not correct and should be rotated by 180°. Could you recheck?

oliv3r, 2022-11-08:
Hey @absalom-muc,

don't think so? I took this picture of you as reference: 
[image] obviously this is the male mounted at the bottom, but the 'tabs' are pointing the same way, board-inside. I agree that the actual connector is on the wrong side and uses the wrong model :p but that shouldn't matter?

Looking at the pcblayout: 
[image]
GND is pin 1 facing the 'middle' +12V is the bottom one facing the edge. As I 'traced' your PCB, I'm fairly confident that the pin orientation is identical to yours?

So pins are GND, MISO, MOSI, CLK, 12V which is exactly what you had as well (though no silkscreen helpers on those labels :p)

oliv3r, 2022-11-09:
Well right now the connector is just a render, and the pinout matches exactly that of a female (the housing is male?) on the bottom. So if you'd solder this board, and put the correct connector on the bottom with the edges facing inwards, your fine.

If I 'flip' the connector to the bottom, it _looks_ like this: 
[image]

But regardless, the pinout stays completly _identical_. So in the end, it's just how the component is being rendered. I do agree it makes logically more sense :p to at least show it exactly as you'd have to assemble it with extension cable type A). But I'm sure excention cable type B exists just as much, which makes it work with the current rendering. Caveat emptor I would say. I'll open a MR with the correct _rendering_ of the connector. But note, that ultimately, the PCB gerber is still the same as before!

oliv3r, 2022-11-09:
https://github.com/absalom-muc/MHI-AC-Ctrl/pull/113

absalom-muc, 2022-11-12:
I understand that the gerber is unchanged and that only the rendering is affected. But I've also seen in the last years that for unexperienced users this minor "inconsistency" creates confusion. Would it be possible to make a female connector and to rotate it by 180°? Then we are consistent to the existing assembly pictures. Thank you.

oliv3r, 2022-11-16:
> Then we are consistent to the existing assembly pictures
I'd have to do some 3D modelling to create that connector :), which is not something I've done before much. So that's certainly a challenge :) So #113 puts it on the correct side, the 180degree rotation would then only be applicable because it's a female header. I'll see what I can do on those rendering skills. We could also omit the render of the connector for the time being :p

absalom-muc, 2022-11-17:
Thank you for your efforts @oliv3r

PR #155 (2023-08-28): KiCad: Fix J1 connector
This fixes the J1 connector footprint and 3D model by using easyeda2kicad [0] to pull the data from LCSC!

Closes #111

[0]: https://github.com/uPesy/easyeda2kicad.py
commit 3419d030 (2023-08-28): KiCad: Fix J1 connector (#155)

This fixes the J1 connector footprint and 3D model by using
easyeda2kicad [0] to pull the data from LCSC!

Closes #111

[0]: https://github.com/uPesy/easyeda2kicad.py
closed 2023-08-28 by absalom-muc