absalom-muc/MHI-AC-Ctrl #112: SRK-35-ZS-W - Status geht, Änderungen allerdings nicht

bruehne, 2022-11-08:
Ich habe Probleme ein SRK-35-ZS-W zu steuern.

Mit iobroker und IoT MQTT Panel habe ich MHI-AC-Ctrl/set/Power On oder Off per MQTT gepublished, allerdings schaltet das Klimagerät nicht. Temperaturänderungen oder die Änderungen der Betriebsmodi klappen auch nicht. Die Konsole zeigt die Änderungen per MQTT an, aber die Steuerbefehle scheinen nicht zu funktionieren.

Die Statuswerte vom Klimagerät werden korrekt ausgelesen und per MQTT gesendet. Ich sehe z. B. die Temperatur, die das Innengerät misst und sehe auch direkt Änderungen, wenn ich mit der Fernbedienung Einstellungen vornehme.

In der Konsole sehe ich nach dem Start OK für die Frequenzen von MISO, MOSI and CLK, allerdings erscheint alle paar Sekunden Error -4 (err_msg_timeout_SCK_high).

Das erste gelötete Board habe ich mit einem 30cm - Kabel angeschlossen. Ich habe auch noch ein zweites Board gelötet und direkt auf den JST-Stecker der Innengerät-Platine gesteckt.

Ich habe auch mal die Testprogramme laufen lassen. Vielleicht gibt es noch weitere Ideen.
[SRK35ZS-W.frequenzen1.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/9965335/SRK35ZS-W.frequenzen1.txt)
[SRK35ZS-W.frequenzen2.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/9965336/SRK35ZS-W.frequenzen2.txt)
[SRK35ZS-W.vcd.txt](https://github.com/absalom-muc/MHI-AC-Ctrl/files/9965337/SRK35ZS-W.vcd.txt)

absalom-muc, 2022-11-12:
Could you provide a log of the serial debug output?
You could try with 160MHz CPU frequency, for some users this works better.
And please use the English language.

absalom-muc, 2023-01-28:
Since there was no feedback from the issue author -> closed. In case of further questions, please feel free to reopen this issue.

closed 2023-01-28 by absalom-muc