absalom-muc/MHI-AC-Ctrl #115: PressureValues, HP, LP are these Values available ?

seepv, 2022-11-23:
Are the PressureValues, HP and LP (25, 26) available for maybe all SRKs or do they get calculated in f.e. the RC-E5 or RC-EX?

absalom-muc, 2022-11-27:
At least my AC indoor unit SRK 35 ZS-S doesn't support the PressureValues, HP and LP (25, 26).

seepv, 2022-11-29:
Related to this Source
https://www.researchgate.net/publication/225309224_Variable_fuzzy_control_for_heat_pump_operation
discharge superheat (Tsh) equals discharge temperature (Td) - condenser temperature (Tc).
Tsh = Td - Tc
or for an SRK:
TDSH = TD - THI-R1.

LP ≙ THI-R1 (Middle of the IndoorCoil/Evaporator in CoolingOperation) and
HP ≙ THI-R1 (Middle of the IndoorCoil/Condensor in HeatingOperation).

THI-R1 is the condenser temperature (in HeatingOperation) and can be converted to HP by one of the following Tables:
https://hvac-eng.com/refrigerants-p-h-diagram/

The Middle of the OutdoorCoil could be THO-R1.
THI-R1 would than analogue HP in CoolingOperation or
LP in HeatingOperation.

glsf91, 2023-10-02:
Closed because of no activity for a long time

closed 2023-10-02 by glsf91