absalom-muc/MHI-AC-Ctrl #118: [Feature request] control low temperatures

Timple, 2022-12-04:
Slightly related to #117

As far as I can tell the airco cannot control to temperatures below 18°C. I would like to control the room temperature to 10°C.

Perhaps this could be implemented by fooling the airco. So if a setpoint of lower than 18°C is given (like 10°C), the airco is supplied with a room temperature which is 'return air temperature' + 8.
That way the airco would effectively control to 10°C.

absalom-muc, 2022-12-05:
I believe you can solve it with your home automation SW communicating with MHI-AC-Ctrl. It sounds for me very specific, therefore I wouldn't implement it in MHI-AC-Ctrl.

Timple, 2022-12-10:
> I believe you can solve it with your home automation SW communicating with MHI-AC-Ctrl.

Actually I cannot, because of #117. I cannot read the temperature and set it as well unfortunately.

absalom-muc, 2022-12-10:
I believe the 18°C was a limitation given by the AC and MHI-Ctrl-AC checks if the setpoint via MQTT is in the valid range. If you want to try it you can replace [row 79 of MHI-AC-Ctrl](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/67ffef767d02bf76571c9f28944cb357ee688892/src/MHI-AC-Ctrl.ino#L79)
`if((f >= 18) & (f <= 30))`
by
`if((f >= 10) & (f <= 30))`
and see if it works w/o the range check.

Not clear why you reference [#117](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/117). 'return air temperature'  is part of the operating data provided by the AC and can't be changed manually as far as I know.

Timple, 2022-12-11:
> Not clear why you reference https://github.com/absalom-muc/MHI-AC-Ctrl/issues/117. 'return air temperature' is part of the operating data provided by the AC and can't be changed manually as far as I know.

That's the thing, it can be changed manually. Because it reports not perse the return air temp, but the external sensor temp.

So if I would implement an offset with home automation, I would typically do: `external_sensor = return_air_temp -10`. However `return_air_temp` is influenced by this, so that does not work.

Or you had a different suggestion in mind when suggesting external home automation?

glsf91, 2022-12-12:
If you have a setpoint of 18 degrees and your room sensor is also 18 degrees and you want to heat up to 16 degrees in your room, then add 2 degrees to your room sensor (=20 degrees) to set Troom.
Airco thinks it is 20 degrees in your room. When the room becomes below 16 degrees (airco thinks it is 18 degrees) it will heat the room until 16 degrees.

Timple, 2022-12-12:
> If you have a setpoint of 18 degrees and your room sensor is also 18 degrees and you want to heat up to 16 degrees in your room, then add 2 degrees to your room sensor (=20 degrees) to set Troom.

For room sensor I use the airco itself. So the above flow will report 20 degrees for return air temp. I will then again add 2 degrees to fool the thermostat. It will then 'measure' 22 degrees. I will again add 2 degrees and it will measure 24 degrees. etc...

Note that your suggestion does work when I would use an external thermostat.

Perhaps I could subtract my own offset again from the measured values, but I'm a bit afraid of race-conditions there. Therefor it would be preferred if this feature could be embedded here. I am okay if it is decided not to implement this, but please understand (or make me understand) the issue completely before denying :slightly_smiling_face:

glsf91, 2022-12-12:
When using the internal room sensor of the airco it is not possible to do this. Because to use the internal sensor a special value has te be used for Troom. So you cannot adjust Troom.

DYLaKo, 2022-12-16:
10 degrees should be possible, since the airco units have the option "Night Setback" on the remote control. The manual explains that when activated, the airco will keep the room temperature at 10 degrees in heating mode (doesn't have a function in cooling mode).

glsf91, 2023-10-02:
Closed because of no activity for a long time

closed 2023-10-02 by glsf91