absalom-muc/MHI-AC-Ctrl #121: SRK-20 ZS-W is Switching off after Router restart

seepv, 2022-12-10:
Thanks for MHI-AC-Ctrl.
With an SRK-20 ZS-W and MHI-AC-Ctrl (2.6R2) the AC is switched off with a router restart.
Is this a wanted behavior or a bug?

absalom-muc, 2022-12-10:
If it takes more than 120 seconds, it is a known behavior described in [Troubleshooting.md](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-ac-switches-power-off-sometimes)

glsf91, 2022-12-16:
With a Fan or setpoint change the AC won't switch on. If you send a set/Power=On, the AC will switch on again.
Maybe you can add this to the documentation.

absalom-muc, 2022-12-21:
> With a Fan or setpoint change the AC won't switch on. If you send a set/Power=On, the AC will switch on again. Maybe you can add this to the documentation.

I don't understand your comment. AC switches off when there is no WIFi for more than 120 seconds. During this time a command can't be received.

glsf91, 2022-12-22:
In Troubleshooting is mentioned:
"You can leave the error state by sending a command via IR-RC or by sending **a command** via SPI"
I read this as sending any command. This is not the case.
When the AC is off because of the 120 seconds timeout, you can switch the AC on again only with the set/Power=On (or remote control). Not with for example changing setpoint. Of course only when wifi or MQTT is back online.

So my suggestion is changing the line in:
"You can leave the error state by sending a command via IR-RC or by sending a Power=On command via SPI"

absalom-muc, 2022-12-22:
The statement in Troubleshooting is correct, you leave the error state (=> errorcode=0) by the given options, but I see you point and added "To switch the AC again on via SPI you have to send the Power On command." to Troubleshooting

closed 2023-03-28 by glsf91

jeroenleenarts, 2024-02-12:
I have installed things on two units, SRK20ZS-WF and a SRK35ZS-WF. Only thing I had to do was unplug the WF-RAC module that comes pre-installed.

(Do not spend the money on the WF version of these MHI units. Their Wifi integration is no match against putting an ESP on the same port.)

sebster6, 2024-06-06:
> (Do not spend the money on the WF version of these MHI units. Their Wifi integration is no match against putting an ESP on the same port.)

Excellent advise! Changed my order from three SRK20ZS-WF to SRK20ZS-W and prepared three MHI-AC-Ctrl modules for installation next month.