absalom-muc/MHI-AC-Ctrl #122: SRK-20 ZS-W WeeklyTimerLight (amber) is permanently off.

seepv, 2022-12-10:
With an SRK-20 ZS-W and MHI-AC-Ctrl (2.6R2) the WeklyTimerLight is permanently switched off.
Is this a wanted behavior or a bug?

WorlockM, 2022-12-12:
I'm not sure if I understand you correctly. But to my knowledge when anything is connected to the SPI port timer functionality is disabled. That's not specific to this project, but also with the Airconwithme or anything else. So also no LED will be on then..

synegic, 2023-02-06:
The AC unit probably expects the device that is connected to the CNS port to handle timers. This is probably to prevent any clashes in control. I bet the official WF-RAC module also handles timers and schedules instead of programming those in the unit itself.

glsf91, 2023-03-28:
Timers are disabled when CNS port is used.

closed 2023-03-28 by glsf91