absalom-muc/MHI-AC-Ctrl #126: Request: “eco mode” and “night set back”

groofoo, 2022-12-17:
So here’s my request and the reasons for the requested.

But first things first.
I would like to thank everyone who made it possible to spend so much time to get this project going and doing the hard work.
I personally use the ESPHome version for the MHI which is forked from this repository.

So my requests….
I would really like the eco mode enabled.
Because (in my use case) it safe’s me about 30% in energy.
With the high energy prices now a day’s, I think a lot more people will be interested in this feature.

And i would like the night set back mode enabled.
Because in the night my rooms do not cool down that much in these cold temperatures (-8 Celsius) we are having right now.
So it is useless to leef the AC units in heating in the night time.

Name of my AC:
Used MHI-AC-Ctrl SRK35 and SRK20

absalom-muc, 2022-12-21:
Since we don't know if / how “eco mode” and “night set back” are coded in the SPI protocol, I'm not too optimistic. Please compare with [issue 77](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/77).

th3sniper86, 2022-12-22:
[image]
The AC just fools you be lowering the set temperature.

absalom-muc, 2022-12-22:
Then I suggest you implement it as part of your home automation system. This gives you additional freedom to set the desired temperature reduction/increase. In general I prefer to limit the functionality of MHI-AC-Ctrl to a minimum and to implement addtional functionality on a higher level.

groofoo, 2022-12-24:
i have tried that in many times the last week.
it seems that i had the sequence of my automations wrong.
old situation
- set climate temperature incl. high and low target temperatures
- set fan mode (high, medium, low, auto)
- set swingmode.
in this order, the energy usage is way higher and fan mode in "auto" keeps blasting the air out in the highest fan mode

new situation
- set swingmode.
- set fan mode "auto"
- set climate temperature incl. high and low target temperatures
i let the room temperature update every 2 minutes and and restart the automation if temperature changes 
some how in this sequence the energy  usage is lower and the fan mode is actually self regulating the fan speeds

and as mentioned, i use the ESPHome version of MHI-AC-Ctrl that is forked from here.
still fine tuning the automations tho.

harm-nedap, 2023-01-03:
@th3sniper86 Humidity is also taken into account, according to the [SCM-SM-315 manual](https://www.mhi-aircond.ru/support/technical-support/technical-library/r32_inverter_2/):
[image]

buennerbernd, 2023-01-04:
@groofoo At least in my devices the eco-mode stays active until I change it with the IR-romote control.
So you cannot control it, but you can use the eco-mode even if you use the MHI-AC-Ctrl.

kARTechnology, 2023-01-26:
> @groofoo At least in my devices the eco-mode stays active until I change it with the IR-romote control. So you cannot control it, but you can use the eco-mode even if you use the MHI-AC-Ctrl.

True, including eco, silent and 3d auto modes doesn't change using the mhi-ac-crtl when set by remote.

groofoo, 2023-01-26:
well, it does not for me. 
i'll quote myself
> I personally use the ESPHome version for the MHI which is forked from this repository.
> 
i guess the esphome version override the settings.

the auto mode for the fan speeds use automation to throttle up, but does not throttle down.
so writing an automation/script to emulate the eco mode is a pain in the ass and very hard to do.

for now i use the good old fashion RM pro and the RM mini's to set the desired modes and use the mhi-ctrl to monitor the settings.

it is not a must, and it was a request and i understand that figuring out or reverse engineering things like this take's up a lot of free time and head scratching.

keep up the good work, and thanks a lot for the work already done to make things like this possible

synegic, 2023-02-06:
I searched for bits in the recently discovered bytes (#77) that may indicate the SPI communication has any knowledge of hi-power/eco/night setback/self-clean modes, but the bytes do not change when toggling those via IR remote control.

This tells me these features are mostly based around playing with temperature inside the unit itself, and not actual features they even cared to code in separately. I suppose they could implement similar functionality in the WiFi controller, just like was suggested here.

@groofoo did you open an issue at the ESPHome project? Maybe we can do something about the behaviour you are describing. Auto fan speed support is not present yet in that version as we still need to upgrade the core library version.

closed 2023-10-02 by glsf91