absalom-muc/MHI-AC-Ctrl #127: Why vanes up/down position with RC - > unknown

glsf91, 2022-12-30:
In the documentation is stated for the Vanes topic: "When the last command was received via the infrared remote control then the Vanes status is unknown and the "?" is published"

I have a SRK-35-ZS AC.
When looking with MHI-AC-Trace I see a difference in the MOSI when using the RC button airflow up/down.
When I remove the lines in MHI-AC-Ctrl-core.cpp:
```
if ((vanestmp & 0x88) == 0) // last vanes update was via IR-RC, so status is not known
        m_cbiStatus->cbiStatusFunction(status_vanes, vanes_unknown);
```
the vanes up/down is shown correctly in the Vanes topic according using the RC.

So why is this set to unknown while it looks like it is working?

Only 1 mode is not shown when using the RC. That is the symbol on the RC without any position sign. This is the symbol which is shown after the swing and before the upper horizontal position.

absalom-muc, 2023-01-02:
I had so far no chance to test it, but I could imagine that this is also related to the behaviour after powering on/off the AC you described [here](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/99#issuecomment-1364545482). Maybe it works only after setting fan Auto.

glsf91, 2023-01-02:
I just tested this after power off mains (230V).
It is working right away. No need to set fan to Auto first.

Maybe you can enable this again the software instead of unknown?

absalom-muc, 2023-01-21:
I can't confirm your observation (at least not for my AC). Once I use the Vanes U/D button on the IR RC the status changes to swing without the lines you mentioned. For my test I used an addtional status topic status_test:
```
      // Test 21.01.2023
      if ((vanestmp & 0x40) != 0) // Vanes status swing
          m_cbiStatus->cbiStatusFunction(status_test, vanes_swing);
      else
          m_cbiStatus->cbiStatusFunction(status_test, (vanestmp & 0x03) + 1);
      // Testend
```
I have the indoor unit SRK 35 ZS-S, no idea why it behaves so differently.

glsf91, 2023-01-22:
I have the SRK 35 ZS-W. Strange if there is a difference.

closed 2023-03-27 by glsf91