absalom-muc/MHI-AC-Ctrl #128: Just an idea, question, no issue.

Onshuis, 2023-01-09:
Would it be possible, to somehow, use an external temperature from another device. Lets say, i have a temperature sensor value from domoticz and use this as the actual temperature. I would like this instead of the DS18x20. Or the possibility to choose what actual temperature to use (built in, DS18x20 or external) The temp now comes from the AC indoor unit that is high up on the wall. A DS18x20 with a wire crossing the living room is not an option. I do however have several temperature sensors around the living room. Or perhaps use this temp value from mqtt to reach the setpoint, instead of the build in temp, or the DS18x20. 

Just a question. Thanks.

dennistd4, 2023-01-10:
It is already possible. Just publish that temperature via mqtt to <device>/set/Troom

Onshuis, 2023-01-11:
Wow i did not know that, now there is a lot i do not know, haha, but it is nice to learn.
So thank you very much, i will try this.

closed 2023-01-11 by Onshuis

WorlockM, 2023-01-20:
I didn't know that either. How would that work? Because I could imagine that it's being overriden by the internal sensor the whole time.

Onshuis, 2023-01-20:
I just resend it every 100 seconds. And put the #define
ROOM_TEMP_MQTT_TIMEOUT on 120 seconds.