absalom-muc/MHI-AC-Ctrl #129: [Feature request] .5 degrees with remote controle

glsf91, 2023-01-14:
@absalom-muc 
How do you think about making it possible to get .5 degrees with the RC in addition to ENHANCED_RESOLUTION ?

I already made some code for it (in MHI-ACCtrl.ino). It is now working as follows:
1. set the setpoint with the RC to the setpoint (=x) you want to add .5 degree and wait >6 sec
(or skip this step if you already on the setpoint where you want to add .5 degree; current setpoint must be n.0 degrees)
2. increase the setpoint with the RC with 1 degree (=x+1)
3. slowly count to 7 (wait 7-10 seconds)
4. decrease the setpoint with the RC with 1 degree (=x)
After about 5 seconds the setpoint on the AC will be x.5 degrees

If the setpoint was already n.5 degrees before, you have to do step 1 first!

This cycle will only work with the RC. Using set/Tsetpoint in between will abort this cycle.
Also doing the same steps with set/Tsetpoint won't add .5 degrees. Of course you can still set the setpoint to y.5 degrees with set/Tsetpoint (if ENHANCED_RESOLUTION is enabled).

How do you think about this ? 
If you want to integrate this I can make a pull request.
If yes, I have a question:
Shall I add an additional ENHANCED_RESOLUTION_RC to enable this or just enable this when ENHANCED_RESOLUTION is enabled?
(ENHANCED_RESOLUTION must be enabled to use this anyway)
 
 This can be handy for people using mainly the RC (like me :-)).

absalom-muc, 2023-01-15:
Hi @glsf91 
it is a nice idea, but to be honest, I wouldn't integrate it to MHI-AC-Ctrl because the use case and the setting seems to be very special. Maybe consider creating an own project for this.
Thank you

glsf91, 2023-01-16:
Ok, I understand.

closed 2023-01-16 by glsf91