absalom-muc/MHI-AC-Ctrl #130: MQTT Connection Failed

Siumba, 2023-01-24:
Yesterday I though about updating my Wemos D1 mini with the newest version. Worst decision ever.

The Wemos gets connected to WiFi but almost instantly drops off, getting in to a boot loop. I've tried several things, downgrading to a lower version, I am using PubSubClient 2.7 but I just have no luck at all.

This is the error log:
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 0
***WIFI NETWORKS -- HIDDEN FOR PRIVACY ***
current BSSID: 74:83:C2:34:61:5A, strongest BSSID: 74:83:C2:34:61:5A
Connecting from bssid:74:83:C2:34:61:5A to bssid:74:83:C2:34:61:5A, channel:4
WIFI_CONNECT_ONGOING
 connected to ***WIFI-SSID-NAME***, IP address: 192.168.1.208 (-57dBm)
WIFI_CONNECT_OK
Attempting MQTT connection... reconnect failed, reason -2, WiFi status: 3

What else can I do?

absalom-muc, 2023-01-24:
Please check Troubleshooting.md especially section [ESP8266 crashes periodically](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-esp8266-crashes-periodically) and show us the result of the frequency measurement.

Siumba, 2023-01-24:
I tried everything in those steps. The weird thing even is that yesterday I did an OTA update, so the device was working and connected to the AC and after that nothing works anymore.

Siumba, 2023-01-24:
I've tried doing some more research. As far as the troubleshooting goes, I have no idea how to read the log while connecting in to the AC, as either USB or through the connector boots it up instantly, missing those initial values. I can only see them when debugging via USB but of course that gives values with 0.

As far as the error, it seems to be some sort of MQTT related issue. Everywhere I come to the same issue, related to the Client ID. Even changing the hostname did not help.

As said before, everything worked so I'm not questioning the hardware at all.