absalom-muc/MHI-AC-Ctrl #131: Get Power usage

HpNoTiQ56, 2023-01-27:
Hi,
I'm trying to monitor power usage (W,KWh or A) but the only value I could gather is TD and CT in A.

First, are TD and CT intensities in A? I've monitored them but TD is about 45 thus can't be in A and CT is too low I think!

Could you tell me if it could be done?

thanks,

dennistd4, 2023-01-28:
The CT value is the right one. Multiply it with your actual voltage and you will have the power in watts. Here it is slightly higher (about 10-40 watt) than the power measured by a Shelly EM, but maybe it's because of the length of the power cable (I use the voltage measured by the Shelly for the calculation).

buennerbernd, 2023-01-28:
I‘m using CT * 230 to calculate the current electrical power. 
If my devices are of, this jumps between 0 and 62W. 
During heating I have peaks up to 1600W but normal are 500 - 900W. 
For power consumption I‘m calculating the integral over the time an get reasonable results.

absalom-muc, 2023-01-28:
btw: In the documentaion and in the source code the TD unit is wrong, TD is the Discharge Pipe Temperature  [°C]

buennerbernd, 2023-01-28:
In my installation I have 3 indoor units and one outdoor unit. All devices are powered by the outdoor unit. That are 4 devices in stand-by plus 3 esps. 
The CT is a value of the outdoor unit for the whole installation. All 3 ESPs are reporting consistent values. Perhaps 0,27A is the minimum step that can be reported. 
I will calculate the power consumption in stand-by for the coming hour…

absalom-muc, 2023-01-28:
o.k. makes sense. The resolution of CT is 14/51A=0,2745A

buennerbernd, 2023-01-28:
So, now I have checked after an hour. My calculated consumption is 0.0133kWh. 
The average power consumption of my whole installation in stand-by is 13.3 Watt. 

@absalom-muc Could you please explain 14/51?

absalom-muc, 2023-01-28:
> @absalom-muc Could you please explain 14/51?

This is the factor to calculate the current from the data byte value, see [here](https://github.com/absalom-muc/MHI-AC-Trace/blob/main/SPI.md#29-ct-a) (just added)

HpNoTiQ56, 2023-01-28:
Thanks a lot for your help and clarification. I'm now ready to follow power consumption !
I didn't think that I would make such reactions!
Keep up the good work !
I Close the issue !

closed 2023-01-28 by HpNoTiQ56