absalom-muc/MHI-AC-Ctrl #133: Question: outdoor fanspeed at low temperature

lwestenberg, 2023-02-09:
Perhaps not really the place to ask as it is not a bug or something in the MHI-AC-Ctrl code but hopefully someone can answer my question here:

Does anybody experience that around 0 degrees or lower the outdoor fan always goes to 8 (max) and stays there. This happens here even when the inside temperature is reached and outdoor humidity is low. Compressor freq is about 28-30.

Model: MHI SRK 50-ZS-W (equipped with ESP running MHI-AC-Ctrl)

synegic, 2023-03-04:
Have you tried simply using the remote on 'normal' settings? E.g. not using the silent or auto mode? I can imagine the unit in general just needs to work harder (move more air through the radiator) to get that last little bit of energy out of the air at temps around 0 degrees. There is not much MHI-AC Ctrl can do about that; outside unit fan speed is determined by the AC itself.

DanDee127, 2023-03-05:
When the outdoor air temperature (models SRK20, 25, 35 : TH3 ; models SRK50, 60 : TH2) is -2°C (models SRK50, 60: 4°C) or lower continues for 30 seconds while the compressor speed is other than 0 rps.

The outdoor fan is stepped up by 2 speed step at each 20 seconds. (Upper limit 8th speed)

Reset conditions
When either of the following conditions is satisfied.
a)The outdoor air temperature (models SRK20, 25, 35 : TH3 ; models SRK50, 60 : TH2) is 0°C
(models SRK50, 60: 6°C) or higher.
b)The compressor speed is 0 rps.

lwestenberg, 2023-03-05:
Wow @DanDee127 thanks for the explanation!

glsf91, 2023-03-28:
@lwestenberg If it is solved please close the issue

lwestenberg, 2023-03-28:
Solved, thanks

closed 2023-03-28 by lwestenberg