absalom-muc/MHI-AC-Ctrl #134: Just some feedback and a litte warning :)

HeadCrash66, 2023-02-10:
Hi,

first of all I love this project. Fantastic work.

I would like to give some feedback about my problems getting it to run on my multisplit configuration.
1x oudoor "SCM 45 ZS-W" and 2x indoor units one "SRK 25 ZS-W and one SRK 35 ZS-W".

The OpData for the outdoor unit is the same on both, but OpData for indoor unit differs. I think that's totally okay.

I had some problems getting MHI-AC-Ctrl to work, first I had a little mistake in defining MQTT topic, but my main problem was different.
I always got a WDT Soft reset/boot loop. I tried diffenet things on the software side including the freqency messurement INO. Nothing helped.

Here are the errors I got:

- frequency was correct (3.2 kHz), boot loop
- frequency was wrong (1.7 kHz), boot loop
- frequency was correct, connect to wifi, boot loop
- frequency was correct, connect to wifi, connect to mqtt, boot loop
- frequency was correct, connect to wifi, connect to mqtt, publich data (Yeah!), boot loop (That happend only ONCE an then never again)
- finally my ESP was unable to connect to Wifi again, fully erased, no change

All of them WITHOUT changing anythin on the hardware! Then I decided to have a more comfortable access to the serial console by connection a second ESP with "esp-link" to RX/TX (Serial2 WIFI) instead of laying on the floor with a laptop ...

As I got 3 WEMOS from china (labeld as "ESP12-F") I was thinking that would be easy. Well I was wrong. "esp-link" also boot-looped on my WEMOS, on all 3 (!). WTF?! I flashed ESPEasy and Tasmoto, just for reference, and they seems to work.
Well taht took me "Nahe einem Nervenzusammenbruch", but okay, I had one older NodeMCU and some naked ESP8266, lets try them.

The NodeMCU just worked fine, no boot loop. Okay, lets go a step further, I desoldered the "ESP12-F" from the WEMOS and replaced it by a naked ESP8266 ... and ... it workes fine. And it is still working!

Long story short, be aware that there are maybe some cheap WEMOS D1 mini with "ESP12-F" which are NOT working for MHI-AC-Ctrl. Don't ask me why, but that's what I found out.

May I try to sniff some SPI packets next to see if "3D Auto" or some other function reflected on SPI as well.

glsf91, 2023-02-11:
> May I try to sniff some SPI packets next to see if "3D Auto" or some other function reflected on SPI as well.

See here #77

melmager, 2023-07-26:
i my case Wifi Problem and D1 mini was a to small Cap on 3.3V on some D1 Mini PCB 
got Spikes on 3.3V during WLAN data transfer
so its a good Idea to add a extra minimal 10uF CAP : 3.3V / GND 
symptom:: Connect to WLAN fail , loss connection

SmOeL-swdev, 2023-08-07:
> Hi,
> 
> first of all I love this project.Fantastic work......
> I had some problems getting MHI-AC-Ctrl to work, first I had a little mistake in defining MQTT topic, but my main problem was different. I always got a WDT Soft reset/boot loop. I tried diffenet things on the software side including the freqency messurement INO. Nothing helped.
> 
> Here are the errors I got:
> 
> * frequency was correct (3.2 kHz), boot loop
> * frequency was wrong (1.7 kHz), boot loop
> * frequency was correct, connect to wifi, boot loop
> * frequency was correct, connect to wifi, connect to mqtt, boot loop
> * frequency was correct, connect to wifi, connect to mqtt, publich data (Yeah!), boot loop (That happend only ONCE an then never again)
> * finally my ESP was unable to connect to Wifi again, fully erased, no change
> 
> All of them WITHOUT changing anythin on the hardware! Then I decided to have a more comfortable access to the serial console by connection a second ESP with "esp-link" to RX/TX (Serial2 WIFI) instead of laying on the floor with a laptop ...
> 
> As I got 3 WEMOS from china (labeld as "ESP12-F") I was thinking that would be easy. Well I was wrong. "esp-link" also boot-looped on my WEMOS, on all 3 (!). WTF?! I flashed ESPEasy and Tasmoto, just for reference, and they seems to work. Well taht took me "Nahe einem Nervenzusammenbruch", but okay, I had one older NodeMCU and some naked ESP8266, lets try them.
> 
> The NodeMCU just worked fine, no boot loop. Okay, lets go a step further, I desoldered the "ESP12-F" from the WEMOS and replaced it by a naked ESP8266 ... and ... it workes fine. And it is still working!
> 
> Long story short, be aware that there are maybe some cheap WEMOS D1 mini with "ESP12-F" which are NOT working for MHI-AC-Ctrl. Don't ask me why, but that's what I found out.

@HeadCrash66 have you solved it already or found a workarround, you might want to look into #151

glsf91, 2023-10-02:
Closed because of no activity for a time

closed 2023-10-02 by glsf91