absalom-muc/MHI-AC-Ctrl #138: Frage: wie bekomme ich den Deckel ab?

ymca72, 2023-03-21:
Hallo,
Vorweg  kein Bug sondern eine Frage:
Ich beschäftige mich gerade sehr intensiv mit deinem Projekt und habe heute eines meiner Splitgeräte geöffnet um mir mal die Leiterplatte anzuschauen um zu wissen was mich erwartet. 
Ich bin aber daran gescheitert, die Abdeckung der Leiterplatte zu öffnen.  Ich wollte am Ende keine rohe Gewalt anwenden um dran zu kommen. Ich denke mal, das es Abdeckung auf der rechten Seite ist wo sehr viel Blech zur Abschirmung drauf ist.  Gibt es da einen Trick die zu öffnen?
Da mein Split sehr dicht rechts an der Wand sitzt, sieht man es kaum. Und das zweite Split sitzt sehr hoch, deswegen habe ich erstmal an diesem üben wollen.

Übrigens,  ich habe 20 PCB bestellt (5 oder 20 hat keinen großen Unterschied gemacht) und würde dann auch welche zum Selbstkostenpreis abgeben.

Danke für die Realisierung des Projektes und Support .

ymca72, 2023-03-21:
Not really. It's more for desinfect and clean. I need advice for open the electric compartment. I have no idea how to open it. The AC is an 2016 zr-35m.

dennistd4, 2023-03-22:
You have to follow that video till 1:34. Then you see a screw on the right side. Unscrew that screw, remove the cover it holds and you will have access to the mainboard.

ymca72, 2023-03-22:
Right , i See in last 24h a lot of YouTube-Videos, that the Single screw remove all. Bit my setup is quite different like that shown in that Videos. I See a lot of screws, a lot of Plastik clips and think in my compartment is the pcb rotated 180° against what i see in YouTube.  This is the reason why i ask.

trombose009, 2023-03-25:
> Hallo, Vorweg kein Bug sondern eine Frage: Ich beschäftige mich gerade sehr intensiv mit deinem Projekt und habe heute eines meiner Splitgeräte geöffnet um mir mal die Leiterplatte anzuschauen um zu [...]

Ich hätte Interesse: svenhandloegten@yahoo.de
Melde dich und sag mir gern wie weit du damit gekommen bist

ymca72, 2023-03-26:
Hi,
i solved the issue to open the electronic compartment but it needs more than one screw in my case (additional 3 screws) and the PCB is indeed 180° reverse than the newer modelsfrom  MHI. Anyway, so I´m now little smarter. Thanks to my AC technican for support.
For all with older models: You must lose the srew on the top, downside and on the left side of the compartment. Then you dismantle the cover in front with by slightly lifting the plastic clips on the left and on the top. If the cover away, you can silde carefully the Main-PCB out of AC. Then you see the needed conector.  This is a little nightmare to dismantle all. Also you must cut one of the cable tie around the ferrite core. The second in this area is reuseable. 
To assemble goes just the same only in reverse. Pay attention to all cables when assembling, that they come back into the correct bushings.

Now I waiting of delivery of all needed parts and then again reopen the AC and install the control. Keeps fingers cross that all will work on the first try.

glsf91, 2023-03-28:
@ymca72 Please close the item if your issue is solved.

ymca72, 2023-03-28:
Thanks for Support. It's sovled. Discription above.

Cheers

closed 2023-03-28 by ymca72