absalom-muc/MHI-AC-Ctrl #139: question.

thomvdv, 2023-03-22:
No bug, a question, can i use the wemos d1 mini V4?

absalom-muc, 2023-03-26:
According to my understanding the only v4 difference is that Micro-USB was replaced by USB-Typ-C => should be no problem using it for MHI-AC-Ctrl

(btw: the title of the issue is not very helpful -> changed it)

thomvdv, 2023-03-26:
> According to my understanding the only v4 difference is that Micro-USB was replaced by USB-Typ-C => should be no problem using it for MHI-AC-Ctrl
> 
> 
> 
> (btw: the title of the issue is not very helpful -> changed it)

Thanks, I will make them in a few months, I'll let you know if the v4 works so you can add it to the intructions! I'm sorry about the title, I'm still learning about GitHub :)

closed 2023-03-27 by glsf91

thomvdv, 2023-05-10:
Hi, i can confirm the V4 does work! The only thing is that the ports are now flipped, You need to assamble it with the usb port on the top instead of the bottom of the board!