absalom-muc/MHI-AC-Ctrl #142: Q: HVAC always shows "disconnected", is it possible to get additional debug logging?

henrykuijpers, 2023-04-01:
It could very well be that there was a mistake with soldering and/or a mistake with a component that was purchased. 

When I click in the whole assembled unit, the esp boots up, it presents the wifi network, etc. Then, when i change the credentials, it is included in my router. When i configure the mqtt setup etc (it would be nice to add a little asterisk to the "friendly name" setting, it totally makes sense that I have to fill it in, but when i just started i just didn't get that, because it was all new to me), the entity would show up in Home Assistant. So far so good.

However, the status page would always say "HVAC (disconnected)" and i seem to be unable to improve that situation. 

It would be very nice to be able to debug this somehow, i.e. by seeing any error messages, or any other indicators on where the problem could be. 

Do you have any suggestions? Or does anyone have experience troubleshooting this? 

I can upload a photo later today of the soldered setup and also list all the parts used etc. Maybe someone can spot a mistake in there.

henrykuijpers, 2023-04-01:
Sorry, i forgot to mention this, the unit I'm trying the first is the SRK-50ZS-W.

glsf91, 2023-04-01:
Take a look with MQTT explorer if the MQTT topics where all there .
I have not HA. Did you also this topic: #29 
You can also debug by connecting an USB cable to the esp and monitor the serial port.

glsf91, 2023-04-16:
@henrykuijpers Is this solved and can this be closed?

henrykuijpers, 2023-04-16:
I'm sorry, i didn't do any debugging anymore. I didn't get it to work with Mitsubishi2mqtt, i did put the ESPHome version on it and then everything started working. So there is a difference somewhere.

 I have 3 other ones i still need to do. Then i can test a little bit more maybe?

glsf91, 2023-04-16:
No problem

SmOeL-swdev, 2023-08-07:
Hi Henry, you are right about the issue you describe that the unit does not update in HA. It requires some adjustments in the MHI-AC-Ctrl firmware to update this. I am figuring this out and hope to solve this asap.

glsf91, 2023-10-02:
Closed because of no activity for a  time

closed 2023-10-02 by glsf91