absalom-muc/MHI-AC-Ctrl #143: Addition to unsupported list

zenstrukt, 2023-04-11:
Providing feedback for the unsupported list. It appears my **SRK71ZEA-S1** does not support the output required. My PCB per the below...

[image]

Unless there's something I'm missing here...

henrykuijpers, 2023-04-16:
Cns is there, but it still needs to be soldered. If you do that, you should be able to use mhi-ac-ctrl.

absalom-muc, 2023-04-17:
> Cns is there, but it still needs to be soldered. If you do that, you should be able to use mhi-ac-ctrl.

Are you sure @henrykuijpers ? The CNS connector of the SRK71ZEA-S1 has 6 pins, but I'm only aware of 5-pin-CNS connectors.

glsf91, 2023-04-24:
Added to not supported list in pull request https://github.com/absalom-muc/MHI-AC-Ctrl/pull/145
Let me know if I have to undo this.

absalom-muc, 2023-08-05:
Thank you for your feedback @zenstrukt

closed 2023-08-05 by absalom-muc