absalom-muc/MHI-AC-Ctrl #151: MISO not zero during startup in software versions 2 and above

NickyNL, 2023-07-22:
Name of my AC: MHI SRK50ZJX-S1
Used MHI-AC-Ctrl HW: 2.2
Used MHI-AC-Ctrl SW: 2.6

I have a strange issue. I have ordered the latest hardware pcb, programmed a ESP with the latest software, and it didn`t work. The ESP keeps getting rebooted due to a too high MISO frequency.
See the serial output here;
Starting MHI-AC-Ctrl v2.6
CPU frequency[Hz]=160000000
ESP.getCoreVersion()=3.1.2
ESP.getSdkVersion()=2.2.2-dev(38a443e)
ESP.checkFlashCRC()=1
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=2400Hz (expected: >3000Hz) out of range!
MOSI frequency=300Hz (expected: <SCK frequency) o.k.
MISO frequency=165Hz (expected: ~0Hz) out of range!

I have tried the following to solve this;
- Try different Wemos D1 versions (standard mini and the PRO), both give the same issue
- Try new level shifters, all the same issue
- Downgrade to software version 2.0, same issue

Lastly I downgraded even ot 1.4, and that seems to at least work.
Now I do get data via MQTT, and the MISO frequency seems to be roughly zero as it should.
I do get some checksum errors, but this at least works.
See the serial output here;
MHI-AC-Ctrl starting
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=2594Hz (expected: >3000Hz) out of range!
MOSI frequency=315Hz (expected: <SCK frequency) o.k.
MISO frequency=0Hz (expected: ~0Hz) o.k.
Found 0 DS18xxx family devices.
Unable to find address for Device 0
Device 0 Address: 0000000000000000
........ connected to xxx, IP address: xxx
OTA Ready, IP address: xxx
Attempting MQTT connection...connected
Wrong MOSI signature: 0x6C 0x80 0x04 or checksum received!

So the main question boils down to, how can the measured MISO frequency be higher when I use the latest software versions?

It`s nice to have it working, but keeping up with the newer versions would be preferrable.

NickyNL, 2023-07-23:
I have tried two more things;

- With software version 2.6, just for testing i disconnected only the MISO pin between the pcb and the airco.
  Doing this resulted in a MISO readout of zero,  a voltage on the MISO pin on the airco side of zero,
  And the data worked only partially. It only send out data from the indoor unit as far as I could see.
- In software version 2.6, I commented out the while(1) so the software wouldn`t reboot, to see if for example MISO would later 
  on become zero.
  After this the program seems to work and I get all the data. Also sending commands works. However the MISO does not go 
  to zero.
  And also the serial output is clogging with "mhi_ac_ctrl_core.loop error: -4"

glsf91, 2023-07-23:
Removing the while(1) can damage the ac or your board as mentioned in Troubleshooting.md
There is no difference in measuring the pins in 1.4 and 2.6.
If the board is connected to your ac with a cable, maybe it is picking up some noise.

absalom-muc, 2023-07-23:
And please re-check your PCB if there is a short between MISO signal and another signal, maybe a bridge due to poor PCB quality or by solder. I suggest a loupe.

NickyNL, 2023-07-23:
Thanks for your responses.
Yes I did read the disclaimer about damaging, I knew it was a bit risky.

Also I did look at #22 which is about the wrong level shifter,
I also looked at #26 and #84, but none seems to display exact behavior to what I am seeing.

So picking up some noise would sound reasonable... However, it does not explain the difference in 1.4 and 2.6.

Also a fault on the PCB would not explain that difference if you ask me.

I have though measured by multimeter,
And the only resistive values I saw where 10k/20k due to the pull/up/down resistors on the level shifter.
Measureable between ground and SCK/MOSI/MISO is some 10k/20k, and between SCK/MOSI/MISO also some 10k/20k.
Between VCC(12V) and SCK/MOSI/MISO it`s all measuring mega ohms.

I do have oscilliscopes and stuff to measure things, but unsure at the moment of what to measure.

glsf91, 2023-07-23:
If you have a logic analyser this will help.
Otherwise maybe with the oscilloscoop you can monitor the behavior of the MISO line at the board without the esp. It should be steady 0 Volt. 
Otherwise strip the program to only measure the frequency repeatly and look with the oscilloscoop.

NickyNL, 2023-07-23:
I don`t have a logic analyser readily available.
When I have some more time I`ll sure hookup a scope to at least look what the signals look like.

I just measured with my multimeter, the MISO pin reads a steady 2,6V when the PCB is connected to the airco. Even without the ESP on it. Without the PCB so measuring just the airco pins it the MISO pin reads 0V.
So at least this voltage must be coming from the on the PCB itself (including level shifter).

Am wondering now if it is due to the resistances on the level shifter.
So measuring with my scope I should be able te verify if it is a voltage diveded copy of SCK and/or MOSI.
Would suspect SCK since that is 5V.
(Although I am wondering why nobody else had the issue)
These are the level shifters I`ve used; https://nl.aliexpress.com/item/1972789887.html

I`ll update next when I`ve had time to hook up a scope!

SmOeL-swdev, 2023-08-04:
@NickyNL have you managed to solve the issue? I was facing the same issue. I have made a complete different hardware version though (manually soldered with stuff I had laying arround).
Initially I used the same pinning as described in the repository. 
I solved the issue by changing the MISO pin to GPIO16 (Seems like internal crosstalk of the ESP). 
I have created al sorts of filtering initially that didn't help, changing pin did. You might need some changing when using the PCB version.

Groeten

NickyNL, 2023-08-04:
@SmOeL-swdev thanks for the tip! Will definitely try swapping to that pin soon.

It also was a good trigger that I still had to do something.
As it is "working" as is, it sometimes gets forgotten.

So, hereby at least some captures with a oscilloscope;
Measured straight on the pins on the pcb coming from the AC.
With the ESP on the pcb as well.

[image]
[image]
[image]

NickyNL, 2023-08-05:
I`ve given your trick a try, and it works!
Swapped MISO from GPIO12 to GPIO16 and now the MISO reading remains 0!
Reimplementing the MISO check on boot and everything is okay now.

Cudos to @SmOeL-swdev!
Of course I live and work in probably the same country, the Netherlands :)

closed 2023-08-05 by NickyNL