absalom-muc/MHI-AC-Ctrl #152: Question: LED state control

nbeernink, 2023-08-12:
Hi,

Just curious if this project can control the room unit LED state/intensity during operation.

I'm guessing there isn't a way since the PCB mimics the SPI commands sent by the IR remote and there aren't any buttons to control the green LED. 

If not we'll just resort to lofi solutions (tape). :)

glsf91, 2023-08-13:
no, this is not possible

nbeernink, 2023-08-13:
Thanks for the quick response!

closed 2023-08-13 by nbeernink

TimCinel, 2023-10-22:
Fellow tape user, it's possible to control LED brightness with the remote for SRK series models. From [the service manual](https://www.manualslib.com/manual/1601239/Mitsubishi-Srk-Series.html):

Use the `MENU` button:
[image]

[image]

Press the timer up/down buttons, and `SET` to confirm. HTH!

(I know this is closed, sorry)

arjanvanderoest, 2025-08-11:
Sorry for bumping this topic, but I can't find a definite answer. I'm eying on ordering the required hardware for this project but having the led OFF (LV0) is a must for me. Since MHI-AC-Ctrl seemingly can't control this: if the remote sets LV0, is this then kept at 0 when sending an 'on' command via SPI? 

Reason for asking: I currently use the Tado Smart AC via Infrared and while the ON command via the remote was set to LV0 the ON command sent via Tado actually set the LED to full brightness again (and sending OFF/ON via the remote again set it to LV0. (Btw: Tado fixed this for me by implementing a different IR command -et in Tado and now via the Tado the led is kept off.).

koosdraadloos, 2025-08-11:
Ok cool, so that setting is retained. Does the same apply for night mode, which is always switched on for me. 

Odd that originally the tado “on” command via IR did an override on what was set via the remote btw.