absalom-muc/MHI-AC-Ctrl #157: Questions regarding protection messages

hvorragend, 2023-09-14:
Outdoor Unit: SCM45ZS-S
Indoor Unit No.1: SRK35ZS-W
Indoor Unit No.2: SRK35ZS-WF

I have a very general question about MHI air conditioning units. This has nothing to do with the project here. But I don't know where else to ask the question.

The second indoor unit was installed just a few weeks ago. Three years later after the other components. 
Then I just noticed the other day that I have contents in the protection state field. I noticed this because my air conditioning suddenly stopped cooling. The technician filled up the refrigerant. But I still get the following different messages very soon after switching on.

Protection modes warnings:
- Low pressure protection control
- Condensation prevention control
- Anti-frost prevention control
- Discharge pipe temperature protection control

The second indoor unit has the MHI WiFi module installed. I don't see any errors in the Smart M-Air app. There are also no LEDs flashing on the indoor units.

Are these normal warning messages or should I be worried? So is this normal or not?

Thanks so much.

absalom-muc, 2023-09-15:
I don't know how it behaves.
I could imagine both. That Protection modes shows current messages or old messages which were not deleted after the error was removed. (I don't know how old errors could be deleted.)
According to my experience the MHI ACs are robust - I would be not worried.

glsf91, 2023-09-23:
I assume solved. I will close this one

closed 2023-09-23 by glsf91