absalom-muc/MHI-AC-Ctrl #158: Can not Connect to wifi on Platform IO and MQTT not working on Arduino IDE if I try to upload with that platform

koalatje2000, 2023-09-19:
Name of my AC:
Used MHI-AC-Ctrl HW:

**Describe the bug**
When I use the Arduino IDE I get an error that PubSubClient is not working because the version is not 2.7 but 2.8 so I switched over to platform IO project. But now I get an error that I can not connect to the wifi. And yeah it is not the password and it showing in the list of wifi names it is show. I was possible to connect on the Arduino IDE but I got there the PubSubClient error about connecting to MQTT but it was connected to the wifi also showing up in my netwerk when I can it. 

I hope someone can help me with this problem.

The result I get from the serial port:
WIFI_CONNECT_ONGOING
WIFI_CONNECT_TIMEOUT
setupWiFi: Start async scanNetworks
WIFI_CONNECT_SCANNING
handleWiFiScanResult(): 11 access points available
 1              Ziggo3109743  1 -91dBm 4C:38:D8:30:86:99 secured 04
 2                     Ziggo  1 -91dBm 5E:38:D8:30:86:99 secured ff
 3           H369AA23E4C_EXT  1 -87dBm 40:3F:8C:A2:9A:C2 secured 08
 4               VFNL-5461B8  1 -92dBm 00:1D:AA:54:61:B8 secured 04
 5   H369AE152E2_EXT 2.4 GHz  1 -90dBm 14:EB:B6:DE:3E:3C secured 04
 6                    Blauw1  2 -77dBm CC:32:E5:43:0E:F8 secured 08
 7             VRV951727B710  6 -66dBm 04:A2:22:27:B7:10 secured 04
 8              ZiggoA5EF24D  6 -94dBm 90:5C:44:C8:A2:55 secured 04
 9                     Blauw 10 -52dBm 9C:5C:8E:8A:39:10 secured 04
10                     Hakjo 11 -81dBm 4C:1B:86:6D:4B:FA secured 04
11                     Hakjo 11 -85dBm 68:FF:7B:8A:91:F5 secured 08
current BSSID: FF:FF:FF:FF:FF:FF, strongest BSSID: 
No matching AP found (maybe hidden SSID), however try to connect.
WIFI_CONNECT_ONGOING
WIFI_CONNECT_TIMEOUT
setupWiFi: Start async scanNetworks
WIFI_CONNECT_SCANNING
handleWiFiScanResult(): 13 access points available
 1              Ziggo3109743  1 -91dBm 4C:38:D8:30:86:99 secured 04
 2                   Messing  3 -82dBm 6C:5A:B0:31:23:52 secured 04
 3             VRV951727B710  6 -66dBm 04:A2:22:27:B7:10 secured 04
 4              Ziggo293E261  6 -85dBm 90:5C:44:46:04:75 secured 04
 5                     Ziggo  6 -86dBm 92:5C:14:46:04:75 secured ff
 6              ZiggoA5EF24D  6 -89dBm 90:5C:44:C8:A2:55 secured 04
 7                     Ziggo  6 -91dBm 92:5C:14:C8:A2:55 secured ff
 8                     Ziggo  1 -91dBm 5E:38:D8:30:86:99 secured ff
 9   H369AE152E2_EXT 2.4 GHz  1 -91dBm 14:EB:B6:DE:3E:3C secured 04
10                     Blauw 10 -44dBm 9C:5C:8E:8A:39:10 secured 04
11           H369AA23E4C_EXT  1 -89dBm 40:3F:8C:A2:9A:C2 secured 08
12                     Hakjo 11 -84dBm 4C:1B:86:6D:4B:FA secured 04
13                     Hakjo 11 -83dBm 68:FF:7B:8A:91:F5 secured 08
current BSSID: FF:FF:FF:FF:FF:FF, strongest BSSID: 
No matching AP found (maybe hidden SSID), however try to connect.
WIFI_CONNECT_ONGOING

glsf91, 2023-09-20:
What is the value of #define WIFI_SSID in support.h ?
No space included by accident?

You can also install in the arduino IDE the pubsubclient 2.7 if you want.

koalatje2000, 2023-09-20:
> What is the value of #define WIFI_SSID in support.h ?
> 
> No space included by accident?
> 
> 
> 
> You can also install in the arduino IDE the pubsubclient 2.7 if you want.

I don't have a space in the name. And I have problems with libraries with Arduino because for other codes I need to use the 2.8 so that is why I just switched to platform io. But I tried but then I get the mqtt error

glsf91, 2023-09-20:
What is the value of #define WIFI_SSID in support.h you used?

koalatje2000, 2023-09-20:
> What is the value of #define WIFI_SSID in support.h you used?

I have the following:
#define WIFI_SSID "Blauw"

I also tried it with a hotspot on my phone to see if it will connect with that

absalom-muc, 2023-09-20:
v2.8 is not supported, see https://github.com/absalom-muc/MHI-AC-Ctrl#software or https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-you-use-pubsubclient-v280

 
 

Gesendet: Mittwoch, 20. September 2023 um 14:30 Uhr

absalom-muc, 2023-09-20:
Sorry, then I misunderstood it. I have never used platformio, so unfortunately no idea. Maybe you could try temporary using the Arduino IDE with v2.7, to check if it is really related to platform IO or another issue.

Message ID: ***@***.***>

glsf91, 2023-09-20:
I'm also using platform IO without problems :-)

Are you 100% sure the wifi password is correct in support.h?

koalatje2000, 2023-09-20:
Yes i am sure it is correct. When home i will try to run it on a fresh system maybe there is some type of conflict in libraries from other projects, but i think not but to be sure.

glsf91, 2023-09-20:
I also I don't think this is the case.
The log line "current BSSID: FF:FF:FF:FF:FF:FF, strongest BSSID:" should show "current BSSID: FF:FF:FF:FF:FF:FF, strongest BSSID: 9C:5C:8E:8A:39:10"
It is looking up the defined WIFI_SSID "Blauw" in the printed list of AP's. So also check if there is not an invisible not printable character in the #define WIFI_SSID "Blauw"
(delete "Blauw" including quotes and type it again in support.h)

koalatje2000, 2023-09-20:
It was something i think with conflict with some library because on a fresh machine it was working the connecting of wifi. But now i have the mqtt error again. (and yeah pubsubclient 2.7.0)

MQTTreconnect(): reconnect_trials=11
 reconnect failed, reason -2, WiFi status: 0
setupWiFi: Start async scanNetworks
WIFI_CONNECT_SCANNING
handleWiFiScanResult(): 5 access points available
 1                    Blauw1  2 -84dBm CC:32:E5:43:0E:F8 secured 08
 2             VRV951727B710  6 -73dBm 04:A2:22:27:B7:10 secured 04
 3                     Blauw 10 -58dBm 9C:5C:8E:8A:39:10 secured 04
 4                     Hakjo 11 -92dBm 4C:1B:86:6D:4B:FA secured 04
 5                     Hakjo 11 -87dBm 68:FF:7B:8A:91:F5 secured 08
current BSSID: CC:32:E5:43:0E:F8, strongest BSSID: CC:32:E5:43:0E:F8
Connecting from bssid:CC:32:E5:43:0E:F8 to bssid:CC:32:E5:43:0E:F8, channel:2
WIFI_CONNECT_ONGOING
 connected to Blauw1, IP address: 192.168.1.75 (-88dBm)
WIFI_CONNECT_OK
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
MQTTreconnect(): reconnect_trials=11
 reconnect failed, reason -2, WiFi status: 0
setupWiFi: Start async scanNetworks
WIFI_CONNECT_SCANNING
handleWiFiScanResult(): 4 access points available
 1                    Blauw1  2 -85dBm CC:32:E5:43:0E:F8 secured 08
 2             VRV951727B710  6 -86dBm 04:A2:22:27:B7:10 secured 04
 3                     Blauw 10 -63dBm 9C:5C:8E:8A:39:10 secured 04
 4                     Hakjo 11 -94dBm 68:FF:7B:8A:91:F5 secured 08
current BSSID: CC:32:E5:43:0E:F8, strongest BSSID: CC:32:E5:43:0E:F8
Connecting from bssid:CC:32:E5:43:0E:F8 to bssid:CC:32:E5:43:0E:F8, channel:2
WIFI_CONNECT_ONGOING

glsf91, 2023-09-20:
Maybe take a look into you MQTT server logs

koalatje2000, 2023-09-20:
> You also changed from Blauw to Blauw1. Maybe that fixed it.
> 
> Take a look into you MQTT server logs

Already changed it back and same is just because that accesspoint was stronger signal.

I use home assistant so can not see the logs or don't know how

glsf91, 2023-09-20:
I think this has to do with your homeassistant MQTT broker (or the used MQTT configuration in support.h)
Try to connect with for example http://mqtt-explorer.com/ with the same connection parameters as you have configured in support.h. 
It is not a library issue.

koalatje2000, 2023-09-20:
I think i found the problem, only no idea why it is not working.

When i try it with the test mosquito broker it is working. I get a connection only spi error that is because currently testing it without connected to the AC unit. See logs:

Starting MHI-AC-Ctrl v2.7R4
CPU frequency[Hz]=80000000
ESP.getCoreVersion()=3.1.2
ESP.getSdkVersion()=2.2.2-dev(38a443e)
ESP.checkFlashCRC()=0
Measure frequency for SCK, MOSI and MISO pin
SCK frequency=0Hz (expected: >3000Hz) out of range!
MOSI frequency=0Hz (expected: <SCK frequency) out of range!
MISO frequency=0Hz (expected: ~0Hz) o.k.
OTA Ready
setupWiFi: Start async scanNetworks
WIFI_CONNECT_SCANNING
handleWiFiScanResult(): 12 access points available
 1              Ziggo3109743  1 -86dBm 4C:38:D8:30:86:99 secured 04
 2                    Blauw1  2 -73dBm CC:32:E5:43:0E:F8 secured 08
 3                   Messing  3 -89dBm 6C:5A:B0:9B:05:EE secured 04
 4                   Messing  3 -83dBm 6C:5A:B0:31:23:52 secured 04
 5               VFNL-5461B8  1 -89dBm 00:1D:AA:54:61:B8 secured 04
 6                     Ziggo  1 -89dBm 5E:38:D8:30:86:99 secured ff
 7             VRV951727B710  6 -56dBm 04:A2:22:27:B7:10 secured 04
 8              Ziggo293E261  6 -86dBm 90:5C:44:46:04:75 secured 04
 9                     Ziggo  6 -89dBm 92:5C:14:46:04:75 secured ff
10                     Blauw 10 -71dBm 9C:5C:8E:8A:39:10 secured 04
11        Galaxy A52s 5GF4A6 11 -91dBm FE:96:F5:95:DD:33 secured 04
12                     Hakjo 11 -89dBm 68:FF:7B:8A:91:F5 secured 08
current BSSID: FF:FF:FF:FF:FF:FF, strongest BSSID: CC:32:E5:43:0E:F8
Connecting from bssid:FF:FF:FF:FF:FF:FF to bssid:CC:32:E5:43:0E:F8, channel:2
WIFI_CONNECT_ONGOING
 connected to Blauw1, IP address: 192.168.1.73 (-79dBm)
WIFI_CONNECT_OK
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-1), WiFi.status()=3 ...
 connected
MQTTclient.connected=1
status=64 topic=connected payload=1
status=64 topic=Version payload=2.7R4
status=64 topic=RSSI payload=-78
status=64 topic=WIFI_LOST payload=0
status=64 topic=MQTT_LOST payload=0
status=64 topic=WIFI_BSSID payload=CC:32:E5:43:0E:F8
MQTTreconnect(): 12 access points available
status=64 topic=fSCK payload=0
status=64 topic=fMOSI payload=0
status=64 topic=fMISO payload=0
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4

But when i try to do it with the my home assistant broker it is not working. Then i get this as result:
current BSSID: CC:32:E5:43:0E:F8, strongest BSSID: CC:32:E5:43:0E:F8
Connecting from bssid:CC:32:E5:43:0E:F8 to bssid:CC:32:E5:43:0E:F8, channel:2
WIFI_CONNECT_ONGOING
 connected to Blauw1, IP address: 192.168.1.73 (-84dBm)
WIFI_CONNECT_OK
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...
 reconnect failed, reason -2, WiFi status: 3
MQTTreconnect(): Attempting MQTT connection (MQTTclient.state=-2), WiFi.status()=3 ...

For the settings i used the following (And yes replaced the password, and ip is the ip of the home assistant):
#define MQTT_SERVER "192.168.1.33"  //"test.mosquitto.org"               // broker name or IP address of the broker
#define MQTT_PORT 1883                              // port number used by the broker
#define MQTT_USER "homeassistant"                                // if authentication is not used, leave it empty
#define MQTT_PASSWORD "PASSWORD"                            // if authentication is not used, leave it empty
#define MQTT_PREFIX HOSTNAME "/"                    // basic prefix used for publishing AC data (e.g. for status),
                                                    // replace "/" by e.g. "/Living-Room/" when you have multiple ACs
#define MQTT_SET_PREFIX MQTT_PREFIX "set/"          // prefix for subscribing set commands, must end with a "/"
#define MQTT_OP_PREFIX MQTT_PREFIX "OpData/"        // prefix for publishing operating data, must end with a "/"
#define MQTT_ERR_OP_PREFIX MQTT_PREFIX "ErrOpData/" // prefix for publishing operating data from last error, must end with a "/"

[image]

glsf91, 2023-09-21:
Try to get it working first with for example http://mqtt-explorer.com/ with the same values of the parameters 
#define MQTT_SERVER "192.168.1.33" //"test.mosquitto.org" // broker name or IP address of the broker
#define MQTT_PORT 1883 // port number used by the broker
#define MQTT_USER "homeassistant" // if authentication is not used, leave it empty
#define MQTT_PASSWORD "PASSWORD" // if authentication is not used, leave it empty

If that won't work, it also don't work with the ESP. The problem is caused by your MQTT broker or the above used values.
I don't have HA so I don't have knowledge about this.

koalatje2000, 2023-09-21:
> Try to get it working first with for example http://mqtt-explorer.com/ with the same values of the parameters 
> 
> #define MQTT_SERVER "192.168.1.33" //"test.mosquitto.org" // broker name or IP address of the broker
> 
> #define MQTT_PORT 1883 // port number used by the broker
> 
> #define MQTT_USER "homeassistant" // if authentication is not used, leave it empty
> 
> #define MQTT_PASSWORD "PASSWORD" // if authentication is not used, leave it empty
> 
> 
> 
> If that won't work, it also don't work with the ESP. The problem is caused by your MQTT broker or the above used values.
> 
> I don't have HA so I don't have knowledge about this. I you use the MQTT add-on maybe this will help: https://github.com/home-assistant/addons/blob/master/mosquitto/DOCS.md

Solved the problem/bug it was not the mqtt part. How I did it was correct. But it has to do with the length of the password. Home assistant automatically creates a password that is 64 characters and that is to long for the pubsubclient. I am not exactly sure what the maximum password length exactly is but is is at least more then 16 characters. If I somewhere can find the maximum length I will mention it later.

To summarise if you can connect to a open broker for example mosquito but not with home assistant it is probably because of the length of the password you have set it to. 

Thx for all the help!

glsf91, 2023-09-21:
The packet send to MQTT in pubsubclient for the connect cannot be more then 128 bytes.
It is not only the password length but also the length of the username (and more like HOSTNAME) which counts.
But with a password of 64 bytes it is going fast.

Please close the issue if it is solved.