absalom-muc/MHI-AC-Ctrl #161: Idea: create separate MHI-AC-Ctrl-Core repo

arpiecodes, 2023-10-11:
Hello!

I'd like to suggest the creation of a separate `MHI-AC-Ctrl-Core` repo where the core library and headers will be stored. This 'core' can then be included/referenced to inside of this repo and the other implementations (such as ESPHome). If it's separated, it's much easier to include it within Arduino projects/not having to copy paste the files over separately.

What do you think?

glsf91, 2023-10-11:
I don't know how absalom-muc thinks about this but for me it is making things more complicated then needed.
If others want to use it, just copy the 2 core files. And for esphome mostly the others files in the esphome repo needs adjustment. So only 1 person have to copy 2 files once in a time :-)
And also for future development (if there is any) it is not defined that the core separation will stay as clean as it is now.

arpiecodes, 2023-10-11:
Fair enough. Would you guys be OK with it if I created such repo myself then? It's a pretty big pain to copy these core files into a HA ESPHome installation by hand all the time for new installs. It'd be much better if we can use the built-in library management feature instead, since updating the core is a pain in the butt there right now.

glsf91, 2023-10-11:
You can do what ever your want in/with your own repo's

arpiecodes, 2023-10-11:
Got it! Thanks!

closed 2023-10-19 by glsf91