absalom-muc/MHI-AC-Ctrl #162: Added inbuilt Webserver, httpupdater, InfluxDB Support, LED status

kARTechnology, 2023-10-18:
I have been using since project since a long time with webserver and influxdb v1 data logging which can be visualised with free chronograf, both are opensource. also the webserver makes it easier to make any schedules using cron. request to please merge and enhance the project as very little changes are necessary. 

the inbuilt led is flashed momentarily when there is data tx/rx from the ac..

output:

[image]

main ino file:
https://www.diffchecker.com/podzoiXc/

changes made in files:  support.h:
```
extern int WIFI_lost;
extern int MQTT_lost;

#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
extern ESP8266WebServer server;
extern ESP8266HTTPUpdateServer httpUpdater;
#include <InfluxDbClient.h>
extern InfluxDBClient* iclient; 
void startHttpServer();
```

changes made in support.cpp:

```
Point point_mhiac("mhi_status");
void buzz(int speed) {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(speed);
  digitalWrite(LED_BUILTIN, LOW);
  delay(speed);
  digitalWrite(LED_BUILTIN, HIGH);
}
```

changes made to function in support.cpp:
```
void output_P(const ACStatus status, PGM_P topic, PGM_P payload) {
  const int mqtt_topic_size = 100;
  char mqtt_topic[mqtt_topic_size];
  Serial.printf_P(PSTR("status=%i topic=%s payload=%s\n"), status, topic, payload);
  if ((status & 0xc0) == type_status)
    strncpy_P(mqtt_topic, PSTR(MQTT_PREFIX), mqtt_topic_size);
  else if ((status & 0xc0) == type_opdata)
    strncpy_P(mqtt_topic, PSTR(MQTT_OP_PREFIX), mqtt_topic_size);
  else if ((status & 0xc0) == type_erropdata)
    strncpy_P(mqtt_topic, PSTR(MQTT_ERR_OP_PREFIX), mqtt_topic_size);
  strncat_P(mqtt_topic, topic, mqtt_topic_size - strlen(mqtt_topic));
  MQTTclient.publish_P(mqtt_topic, payload, true);

  if (!isdigit(String(payload)[0]))
    point_mhiac.addField(String(mqtt_topic), String(payload));
  else {
    String dat = String(payload);
    point_mhiac.addField(String(mqtt_topic), dat.toFloat());
  }
  if (!iclient->writePoint(point_mhiac)) Serial.print(F("InfluxDB write failed: "));
  point_mhiac.clearFields();
  buzz(15);
}
 
```

new http.ino
```

void startHttpServer() {
[... 173 lines cut ...]
      if (SPIFFS.exists(pathWithGz))     // version of that file must be deleted (if it exists)
        SPIFFS.remove(pathWithGz);
    }
    Serial.print("handleFileUpload Name: ");
    Serial.println(path);
    fsUploadFile = SPIFFS.open(path, "w");  // Open the file for writing in SPIFFS (create if it doesn't exist)
    path = String();
  } else if (upload.status == UPLOAD_FILE_WRITE) {
    if (fsUploadFile)
      fsUploadFile.write(upload.buf, upload.currentSize);  // Write the received bytes to the file
  } else if (upload.status == UPLOAD_FILE_END) {
    if (fsUploadFile) {      // If the file was successfully created
      fsUploadFile.close();  // Close the file again
      Serial.print("handleFileUpload Size: ");
      Serial.println(upload.totalSize);
      server.send(200, "text/plain", "Upload Success");

      //    server.sendHeader("Location", "/success.html");     // Redirect the client to the success page
      //    server.send(303);
    } else {
      server.send(500, "text/plain", "500: couldn't create file");
    }
  }
}
```

glsf91, 2023-10-19:
Looks nice but I'm not going to integrate this. I (and also absalom-muc I think) want to keep this module as clean as possible. Too much not related stuff will be added. 
You can make your own fork if you want.

glsf91, 2023-11-06:
I will close this one.