absalom-muc/MHI-AC-Ctrl #165: Unable to send commands to the MHI-AC-Ctrl

Allinijhof, 2023-11-19:
Name of my AC: SRK 35 ZS-W
Used MHI-AC-Ctrl HW: ESP8266 ((WEMOS D1 R2 & mini with 160 MHz)
Used MHI-AC-Ctrl SW: MHI-AC-Ctrl-2.8

I am able to connect the MHI-AC-Ctrl to my aircon, OpData has been displayed successful, updates have been processed through MQTT (ie RETURN-AIR). Unfortunately I am not able to send commands to the MHI-AC-Ctrl. 

Could you please advise what I have to do to send ie Power on to my device?

How can I send other commands like Mode Cool to the device?

Many thanks for your support!

Regards,
Allard

[MHI-AC-Ctrl.zip](https://github.com/absalom-muc/MHI-AC-Ctrl/files/13404596/MHI-AC-Ctrl.zip)

absalom-muc, 2023-11-19:
Have you checked cmd_received (see [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#mqtt-status))?
If you see no reaction on cmd_received then you use the wrong set-topic or something goes wrong with the connection of the MISO signal between ESP8266 board and your AC.

Allinijhof, 2023-11-19:
Hi Absalom,

I have received cmd_received = "unknown command"

absalom-muc, 2023-11-19:
Very good, use e.g. MQTT-Explorer to check the set topic, then you should see what is wrong with your commands. E.g. when switching mode to Cool you should see
[image]

Allinijhof, 2023-11-19:
Hi Absalom, 

I have been reverting the HOSTNAME to default = 'MHI-AC-Ctrl", after sending the command to topic = MHI-AC-Ctrl/set/Power and with message = On the airco started to work.

Apparently the HOSTNAME hasn't been updated everywhere... I changed HOSTNAME in Support.h, where did I miss the update when I wanted to make the controller room specific?

absalom-muc, 2023-11-19:
It should be sufficient changing HOSTNAME only. Maybe recheck upper/lowercase of your changed HOSTNAME

Allinijhof, 2023-11-22:
Spot-on! Thanks four your support!

glsf91, 2023-11-23:
I close this one because it is solved.

closed 2023-11-23 by glsf91