absalom-muc/MHI-AC-Ctrl #166: Implement Vacant Property Mode

tomm1ed, 2023-11-28:
Newer MHI AC's (2022+?) with the Smart M-Air WiFi module allow setting up the Vacant Property Mode which allows for:
Cooling: 31°C to 33°C (in steps of 1°C)
Heating: 10°C to 17°C (in steps van 1°C)
See https://www.mhi-mth.co.jp/en/products/pdf/Smart_M-Air_AppManual-English(EEA,UK).pdf page 20

Is it possible to implement this for MHI-AC-CTRL as well?

glsf91, 2023-11-28:
I expect that this would be possible. Of course depending how the WR-RAC module is communicating this to the AC.
But I don't have this option in my app so I cannot implement this. My AC is from 2022 according to the sticker.

tomm1ed, 2023-11-30:
> I expect that this would be possible. Of course depending how the WR-RAC module is communicating this to the AC. But I don't have this option in my app so I cannot implement this. My AC is from 2022 according to the sticker.

You looking at the Tweakers topic it appears there is some sort of change during 2022 (a firmware change maybe?) which enables this option. There are people with 2022 Airco's with and without this option

closed 2023-12-05 by tomm1ed