absalom-muc/MHI-AC-Ctrl #167: KiCAD: J1 Connector pinout is wront

AlexCPU, 2023-11-28:
The KiCAD design was "fixed" in #155 to model a female connector correctly on the reverse of the board, which involved flipping the connector.  However the schematic wasn't updated, so the PCB layout has the pins defined incorrectly:
[image]
You can see the unconnected nets, and the fact that different named nets are connected to the pins (easiest to see on the GND and +12V pins).
I'll see if I can raise a PR for this.

Unrelated question: I'm making a new version of the board with an ESP32-C3 integrated (i.e. no extra D1 Mini) with mostly SMD parts, any preference in the repo where to put it?

glsf91, 2023-11-28:
That epshome fork is using a part of the mhi-ac-ctrl code.
I have also tried with that esphome fork without success.
So I suggest you first trying to get this up and running without errors before making a new version of the board :-)

closed 2024-01-08 by absalom-muc

oliv3r, 2024-02-07:
Sorry for not noticing this !! Thanks @AlexCPU for the fix :)

PR #168 (2024-01-08): KiCad: Correct J1 connector pinout
Fixes #167
Regenerated PDF and gerber/drill
commit 47d5ca86 (2024-01-08): Correct J1 connector pinout (#168)

Fixes #167
Regenerated PDF and gerber/drill